OpenTracksApp/OpenTracks #426: Failed to scan for Bluetooth devices

sjvudp, 2020-09-10:
**Describe the bug**
(Originally filed by mistake as https://github.com/OpenTracks/OpenTracks/issues/1)
On a Huawei P10 Lite (Android 8.0) I had Bluetooth and location services enabled, and one app was recording a route while also recording the heart rate from a Bluetooth device. I had set up OpenTracks to record the route in parallel, too, and I wanted it to record the heart rate, too.
However when trying to add the heart rate device, I only got an error message (sorry, I didn't save it) that the app cannot scan for Bluetooth devices.
Maybe I should mention that the German "Corona Warn App" was active, too. That could have implications, but I don't really know.
The version being used was 3.9.4.

**To Reproduce**
1. Go to 'Settings/Bluetooth'
2. Click on 'Heart Rate'
3. See error

It seems the problem only occurs if a heart rate sensor is already paired.

**Technical information**
 - Device: Huawei P10 Lite
 - OS: Android
 - Version 8.0

dennisguse, 2020-09-12:
No clue at the moment and I don't have a heart rate sensor at hand.
What you could try: install OpenTracks twice (Fdroid and Playstore) and check if you can connect from one as well as try to search for the sensor in the other.
But in general: it is the responsibility of the operating system.

dennisguse, 2020-09-28:
I just tested with a heart rate sensor using two instances of OpenTracks on one device:

1. While recording in 1st instance (with heartrate), the 2nd does not find the sensor (via settings).
  I guess, it is not announcing it's presence while reading heartrate.
2. When the _same_ heartrate sensor is selected on both devices, recording in parallel on _both_ instances is possible.
  Even start/stop of one instance does not affect the other.

NOTE: Testing done without Corona Warn App as it requires Google Play Services

@sjvudp So, I cannot reproduce your issue.
Please re-open and add more information if the issue is still affecting you.

closed 2020-09-28 by dennisguse

sjvudp, 2020-09-30:
OK, there may be two factors that influence the issue:

1.  The Corona Warn App

1. The "other app" which actually was https://play.google.com/store/apps/details?id=wmdev.apps.navBrowser&hl=en

I don't know whether the second one needs Google Play Services as well, but at least the base app does not require $$$ to try (I purchased the Professional license and a map). I'm not sure the App is fully localized for non-German, but maybe you can try to use the heart rate sensor with Ape Map first, then with Open Tracks. You would start to record a track in Ape Map (see image).

[image]

dennisguse, 2020-10-01:
I cannot reproduce the issue (Pixel2, Android11 default ROM).
First of all I cannot install the app as it is incompatible (don't ask me why).

What I did (same device):
* install OpenTracks from PlayStore
* install OpenTracks from F-Droid

While the heartrate sensor is on (not connected), both installations can search for it and show it in the settings.
When the sensor is reading heartrate (i.e., something connected to it), it does not announce it's pairing information and cannot be searched for anymore.
But both OpenTracks installations show the heartrate correctly.
The Corona Warn App had no effect.

Could you send us a video showing the issue?