OpenTracksApp/OpenTracks #429: OpenTracks doesn't get and record gps data even if allowed

MarcoDjallo, 2020-09-12:
Hello, I recently updated to Android Q and now I have this issue with OpenTracks.

The app neither get or record GPS data from my phone.

I allowed the app to access location; I disabled battery optimization; other apps works fine with GPS.

OpenTracks' notification says only that it is starting the GPS and remains that way. No location access icon appears in my statusbar and in fact android location settings says that OpenTracks never asked for my location, even if it is allowed to do so.

One strange thing I noticed is that, differently from similar apps, when OpenTracks asks me to allow it to receive my location, I can only say "Allow only while using the app", while with other similar apps I also have the option "Allow always".

How can I share more useful infos?
Thank you

**Technical information**
 - Device: Nexus 5X
 - OS: AOSPExtended
 - Version 7.3.1 (Android 10 Q, august patches)

dennisguse, 2020-09-12:
Sounds like the issue described here: https://github.com/OpenTracksApp/OpenTracks/issues/282

But we never fixed anything regarding to that, because we did not got any further information.
PS: See here: https://github.com/OpenTracksApp/OpenTracks/issues/282#issuecomment-646692317

MarcoDjallo, 2020-09-13:
> 
> 
> Sounds like the issue described here: #282
> 
> But we never fixed anything regarding to that, because we did not got any further information.
> PS: See here: [#282 (comment)](https://github.com/OpenTracksApp/OpenTracks/issues/282#issuecomment-646692317)

so there is anything I can do to help solve this issue? In the comments I see that someone mentions Fitotrack, but neither this one works for me. The only one that seems to correctly request my position (location icon appears in the statusbar) is OSMtracker.

Thanks

rgmf, 2020-09-13:
> One strange thing I noticed is that, differently from similar apps, when OpenTracks asks me to allow it to receive my location, I can only say "Allow only while using the app", while with other similar apps I also have the option "Allow always".

If FitoTrack doesn't work for you neither then this "strange thing" doesn't matter for your issue because FitoTrack ask you for "Allow Always".

I don't know if you have Android Studio installed and you use it, but if you use it then you can connect your phone and copy/paste logs here so maybe we can figure out what happen. The only handicap is that you need to record logs in a place where you get GPS signal...

Does a previous OpenTracks version work for you?

MarcoDjallo, 2020-09-17:
Issue solved. 

Was a problem with the gps on my phone, but happened only in some apps. After investigating the issue, I tried to remove and reinsert the sim card and the gps started working again in every app, OpenTracks included.

I'm sorry, didn't seem this way in the first place!

closed 2020-09-17 by MarcoDjallo

dennisguse, 2020-09-17:
@MarcoDjallo Thanks for the feedback!
And cool that it is working ;)