OpenTracksApp/OpenTracks #439: Export fails on Android 11

LangeSW, 2020-09-24:
Version v3.10.0 ob Android 11

When trying to Export all tracks, after selecting a folder and granting access, only one track is exported and the App then seems to Crash.

Another minor issue is, when exporting a second time, the folder is shown, but the button at the bottom "use this folder for Export" is missing ☹️

dennisguse, 2020-09-24:
It sounds like an OS issue that the permission to write into this folder is revoked (by the OS?) while the export is running.
And we do not catch this exception, resulting in a crash.

Can you:
* post a logcat output
* clarify which OS version/variant you are using 
* check that one track is always exported (i.e., export into an empty folder)

As a workaround: can you try to use the "share a track"-functionality from the track list menu?
It also works for multiple tracks, and you could, for example, create an email and then later export it from there.

PS: Thanks for the report.

SaraSmiseth, 2020-09-25:
The button "use this folder for Export" was missing for me as well after trying it once. I navigated into a different folder and then it was there again. It does not export anything for me.

I have a Pixel 3a with stock android 11 Security updates 2020-09-05.

It looks like its trying to read all the folders inside the selected folder and that fails.

```
09-25 13:07:42.627  1585  1790 I ActivityTaskManager: Displayed de.dennisguse.opentracks/.io.file.exporter.ExportActivity: +132ms
09-25 13:07:42.630   622   622 E SELinux : avc:  denied  { find } for pid=3107 uid=10117 name=tethering scontext=u:r:grilservice_app:s0:c117,c256,c512,c768 tcontext=u:object_r:tethering_service:s0 tclass=service_manager permissive=0
09-25 13:07:42.659  6044  6062 E DatabaseUtils: Writing exception to parcel
09-25 13:07:42.659  6044  6062 E DatabaseUtils: java.lang.SecurityException: Permission Denial: reading com.android.externalstorage.ExternalStorageProvider uri content://com.android.externalstorage.documents/tree/primary%3A/document/primary%3ADownload from pid=11612, uid=10224 requires that you obtain access using ACTION_OPEN_DOCUMENT or related APIs
09-25 13:07:42.659  6044  6062 E DatabaseUtils: 	at android.content.ContentProvider.enforceReadPermissionInner(ContentProvider.java:820)
09-25 13:07:42.659  6044  6062 E DatabaseUtils: 	at com.android.externalstorage.ExternalStorageProvider.enforceReadPermissionInner(ExternalStorageProvider.java:147)
09-25 13:07:42.659  6044  6062 E DatabaseUtils: 	at android.content.ContentProvider$Transport.enforceReadPermission(ContentProvider.java:684)
09-25 13:07:42.659  6044  6062 E DatabaseUtils: 	at android.content.ContentProvider$Transport.query(ContentProvider.java:239)
09-25 13:07:42.659  6044  6062 E DatabaseUtils: 	at android.content.ContentProviderNative.onTransact(ContentProviderNative.java:106)
09-25 13:07:42.659  6044  6062 E DatabaseUtils: 	at android.os.Binder.execTransactInternal(Binder.java:1154)
09-25 13:07:42.659  6044  6062 E DatabaseUtils: 	at android.os.Binder.execTransact(Binder.java:1123)
09-25 13:07:42.659 11612 11668 W DocumentFile: Failed query: java.lang.SecurityException: Permission Denial: reading com.android.externalstorage.ExternalStorageProvider uri content://com.android.externalstorage.documents/tree/primary%3A/document/primary%3ADownload from pid=11612, uid=10224 requires that you obtain access using ACTION_OPEN_DOCUMENT or related APIs
09-25 13:07:42.660  6044  6062 E DatabaseUtils: Writing exception to parcel
09-25 13:07:42.660  6044  6062 E DatabaseUtils: java.lang.SecurityException: Permission Denial: reading com.android.externalstorage.ExternalStorageProvider uri content://com.android.externalstorage.documents/tree/primary%3A/document/primary%3ADCIM from pid=11612, uid=10224 requires that you obtain access using ACTION_OPEN_DOCUMENT or related APIs
09-25 13:07:42.660  6044  6062 E DatabaseUtils: 	at android.content.ContentProvider.enforceReadPermissionInner(ContentProvider.java:820)
09-25 13:07:42.660  6044  6062 E DatabaseUtils: 	at com.android.externalstorage.ExternalStorageProvider.enforceReadPermissionInner(ExternalStorageProvider.java:147)
09-25 13:07:42.660  6044  6062 E DatabaseUtils: 	at android.content.ContentProvider$Transport.enforceReadPermission(ContentProvider.java:684)
09-25 13:07:42.660  6044  6062 E DatabaseUtils: 	at android.content.ContentProvider$Transport.query(ContentProvider.java:239)
09-25 13:07:42.660  6044  6062 E DatabaseUtils: 	at android.content.ContentProviderNative.onTransact(ContentProviderNative.java:106)
09-25 13:07:42.660  6044  6062 E DatabaseUtils: 	at android.os.Binder.execTransactInternal(Binder.java:1154)
09-25 13:07:42.660  6044  6062 E DatabaseUtils: 	at android.os.Binder.execTransact(Binder.java:1123)
09-25 13:07:42.660 11612 11668 W DocumentFile: Failed query: java.lang.SecurityException: Permission Denial: reading com.android.externalstorage.ExternalStorageProvider uri content://com.android.externalstorage.documents/tree/primary%3A/document/primary%3ADCIM from pid=11612, uid=10224 requires that you obtain access using ACTION_OPEN_DOCUMENT or related APIs
09-25 13:07:42.661  6044  6062 E DatabaseUtils: Writing exception to parcel
09-25 13:07:42.661  6044  6062 E DatabaseUtils: java.lang.SecurityException: Permission Denial: reading com.android.externalstorage.ExternalStorageProvider uri content://com.android.externalstorage.documents/tree/primary%3A/document/primary%3Apanoramas from pid=11612, uid=10224 requires that you obtain access using ACTION_OPEN_DOCUMENT or related APIs
09-25 13:07:42.661  6044  6062 E DatabaseUtils: 	at android.content.ContentProvider.enforceReadPermissionInner(ContentProvider.java:820)
09-25 13:07:42.661  6044  6062 E DatabaseUtils: 	at com.android.externalstorage.ExternalStorageProvider.enforceReadPermissionInner(ExternalStorageProvider.java:147)
09-25 13:07:42.661  6044  6062 E DatabaseUtils: 	at android.content.ContentProvider$Transport.enforceReadPermission(ContentProvider.java:684)
09-25 13:07:42.661  6044  6062 E DatabaseUtils: 	at android.content.ContentProvider$Transport.query(ContentProvider.java:239)
09-25 13:07:42.661  6044  6062 E DatabaseUtils: 	at android.content.ContentProviderNative.onTransact(ContentProviderNative.java:106)
09-25 13:07:42.661  6044  6062 E DatabaseUtils: 	at android.os.Binder.execTransactInternal(Binder.java:1154)
09-25 13:07:42.661  6044  6062 E DatabaseUtils: 	at android.os.Binder.execTransact(Binder.java:1123)
[... cut ...]
09-25 13:07:42.685 11612 11668 E AndroidRuntime: 	at de.dennisguse.opentracks.fragments.ExportProgressDialogFragment$ExportThread.exportTrack(ExportProgressDialogFragment.java:219)
09-25 13:07:42.685 11612 11668 E AndroidRuntime: 	at de.dennisguse.opentracks.fragments.ExportProgressDialogFragment$ExportThread.run(ExportProgressDialogFragment.java:200)
09-25 13:07:42.689  1585 11670 I DropBoxManagerService: add tag=data_app_crash isTagEnabled=true flags=0x2
09-25 13:07:42.689  1585  9068 W ActivityTaskManager:   Force finishing activity de.dennisguse.opentracks/.io.file.exporter.ExportActivity
```

dennisguse, 2020-09-26:
I can reproduce the issue also on an Android 11.
I will check what we need to do here.

dennisguse, 2020-09-26:
Autsch - this one was interesting.
To write into a file, one now needs write permissions (from Android 11 onwards) - who could have thought that.
So far, read permissions were sufficient (doesn't make any sense).
And I thought "implementation done", when i reached the "works for me state" :-1:

commit bc04efba (2020-09-26): Bugfix: for export all write permissions are required.
Android 11 enforces this (and it is good practice).
Fixes #439.
closed 2020-09-26 by dennisguse