OpenTracksApp/OpenTracks #442: Custom storage location

hockeymikey, 2020-09-29:
Add a setting to allow for browsing for a custom location to store sessions. Biggest thing is this helps with portability. I wanna be able to have this synced up or backed up on my sdcard so I don't have to worry about losing anything between wipes of my device.

dennisguse, 2020-09-30:
I am not entirely sure, but are you requesting an automated export of the tracks to the file system (aka SD card)?

If yes, we are working on that as part of synchronization (see #90).
Some progress regarding automated export was done in #383 by @thrillfall, but it is still work on progress.

hockeymikey, 2020-09-30:
> I am not entirely sure, but are you requesting an automated export of the tracks to the file system (aka SD card)?
> 
> If yes, we are working on that as part of synchronization (see #90).
> Some progress regarding automated export was done in #383 by @thrillfall, but it is still work on progress.

Maybe, I wasn't sure how you are currently storing it. Figured sqlite or something but didn't have the time to check. I'm saying using the sdcard for the storing of the application data. Using the app specific storage which would be /sdcard/Android/data/something.opentracks which is returned from GetExternalFilesDirs() I think.

dennisguse, 2020-09-30:
Yes, we are using a sqlite database and it is stored in the app-private storage.
The photos are stored in `getExternalFilesDir()` (wherever Android thinks this should be stored), but even that is app-private.

At the moment, you need root on your device to access this _raw_ data as Android will prevent access.
As discussed in https://github.com/OpenTracksApp/OpenTracks/issues/346 we cannot even enable the backup via ADB as otherwise backups may be sent to Google's Cloud Storage (not cool).

So, your only chance at the moment (beyond root access) is the 'export all'-functionality (use KMZ to contain everything); and you have to trigger it manually.
Settings cannot be exported.

Long story short: if you need access to the _raw_ internal data, we don't have this functionality.
It can be implemented (i.e., just copy the database from within OpenTracks to some public space) and if you think it is valuable feel free to create a PR.

hockeymikey, 2020-10-01:
> Yes, we are using a sqlite database and it is stored in the app-private storage.
> The photos are stored in `getExternalFilesDir()` (wherever Android thinks this should be stored), but even that is app-private.

That's the point. It's stored on the sdcard then at that point making it modular across devices then very easily. Also, why aren't you using getExternalMediaDirs? That makes more sense.

> At the moment, you need root on your device to access this _raw_ data as Android will prevent access.

Well the first thing is that is will always be accessible by the app, allowing for persistant data across wipes/moving devices. And even if it does need root, some of us have it. Plus the rom I'm working on I'm adding functionality to access that data without root (eventually, it's on the to-do list).

> As discussed in #346 we cannot even enable the backup via ADB as otherwise backups may be sent to Google's Cloud Storage (not cool).

Why does that even matter? If you're worried about Google, get google off your phone. Go install MicroG, or go GApps-less. I'm using Microg and works great. Also this needs to be fixed because some roms (like mine) include Seedvault which is an opensource backup solution allowing for backing up to anywhere like your own Nextcloud, sdcard, flash drive, etc. 

> Settings cannot be exported.

Why. Is there an issue opened?

dennisguse, 2020-10-01:
We store all data within the app-private directories, so that other applications have NO access to any data belonging to OpenTracks.
As of today a lot of apps are have read access to external storage and read whatever they want there.
As privacy is a primary feature, we cannot put there any data (by default).
If the user decides to export the data onto _public_ external storage, then we do it - because the decision was made by the user.
But I am not going to make this decision for them.

PS: The move to Scoped Storage on Android 11 is only enforced for targetSDK 30+.
So, every "older" app can still access the external storage.
And this situation will remain this way for many years as Android's upgrades policy is a bad joke.
As of today we still see see Android 5 devices...

Anyhow: how can we solve your problem without potentially comprising privacy for some users?

hockeymikey, 2020-10-01:
> We store all data within the app-private directories, so that other applications have NO access to any data belonging to OpenTracks.

Maybe I want other apps to have access to that data like my pictures like my gallery app. It frankly should be an option. If you have apps you don't trust, you should be isolating them in say Shelter or uninstalling them. 

> As of today a lot of apps are have read access to external storage and read whatever they want there.
> As privacy is a primary feature, we cannot put there any data (by default).

Again, if you don't trust someone in your house, maybe you need to kick them out. And it wouldn't be default functionality. Not every phone has an sdcard so internal would be default. 

> If the user decides to export the data onto _public_ external storage, then we do it - because the decision was made by the user.
> But I am not going to make this decision for them.

Strawman, nobody ever said for you to make it, just add the option.

> PS: The move to Scoped Storage on Android 11 is only enforced for targetSDK 30+.
> So, every "older" app can still access the external storage.
> And this situation will remain this way for many years as Android's upgrades policy is a bad joke.
> As of today we still see see Android 5 devices...

It's about Google is a joke. Every update of theirs is worse than the rest. It started going downhill in Kitkat. I'm never using anything above Oreo. Pie+ is a complete mess. APIs become garbage, UI is junk. Nothing redeemable. Maybe a good feature or too, but can be backported. 

> Anyhow: how can we solve your problem without potentially comprising privacy for some users?

It's a setting users can change. It's about freedom and by not adding this feature you are telling the user what is right and making that decision for them.

dennisguse, 2020-10-01:
Back in 2019 I had a similar discussion about adding a map with @pstorch, which I did not want to include as it would require Internet access.
In the end, we were building an API and he build OSMDashboard.
Worked like charm.

If you want, build it - if it is technically and from a privacy-perspective sound we will merge it.
If you have a concept how to do it and some feeling for the feasibility, we can continue this discussion.

hockeymikey, 2020-10-01:
@dennisguse I will build it but I'm not wasting my time and merging it to people that lack vision and lazy. Nothing that I have said is hard nor anti-privacy. You think it is, but the fact is it isn't. You don't rebuttal anything because there isn't a rebuttal. You just hate the user and don't want to provide a better experience for them.

dennisguse, 2020-10-30:
Closing for now.

closed 2020-10-30 by dennisguse