OpenTracksApp/OpenTracks #443: Interval Stats announced incorrectly

slackline, 2020-10-01:
**Describe the bug**

I'm liking the lap timer but have noticed recently its off the mark when making announcements.  Yesterday In was out running and logged the following, but whilst running several were announced at 6mXXs (the 6 and 7km laps I think from memory).

[image]

Don't know if its related to #406

**To Reproduce**
1. Record an activity.
2. Note the announced Lap Times, some are accurate others are not.

Can't discern a particular pattern or cause so difficult to reproduce.  I can supply the GPX to the above if it would be of use.

**Technical information**
 - Device: Motorola G7 Plus
 - OS: LineageOS 16.1
 - Version v3.11.0-b3117e641

rgmf, 2020-10-01:
I'll check it out. So the intervals are good but not the announcement, right?

slackline, 2020-10-01:
Yes, as recorded in the above view they are correct it was the announcements that were off.  Thanks for looking into this.

rgmf, 2020-10-05:
@slackline, bug catched... The last interval wasn't computing well and somtimes it failed. I'm going to push a PR.

Thank you ;)

PS/ A couple months ago I installed Lineage OS in my phone withouth Google Services and I cannot use TTS so I only can test all this in a virtual device. Don't hesitate and open this once again if you test it and fail for you please.

closed 2020-10-05 by rgmf

slackline, 2020-10-05:
@rgmf Great work tracking that down and solving it and not a problem on reporting it.  Will await the next build and try it out (currently running 10km every other day and due to go out tomorrow morning at 06:30!).

Thanks for yours and  @dennisguse and anyone else who contributes to the code continued efforts developming and improving OpenTracks.

dennisguse, 2020-10-05:
@rgmf Why not trigger a new release?
Any objections?

rgmf, 2020-10-05:
@dennisguse, a new release will be welcome for people using announcement. This is a big bug for people using it.

slackline, 2020-10-13:
Reporting back on this as I've now had a chance to try it on a couple of runs and spoken lap times whilst out recording seem to be much more accurate.  Not that I stop to check, but I know for instance when I'm running closer to 5m/km as its the same route I follow and I'm now not having any reported as >6m/km which was what flagged up the problem to me in the first instance.  I'll continue to keep an eye on it though.

Great work and rapid fixing, thanks to all involved.

commit a036a339 (2020-10-05): Bugfix: interval stats announced incorrectly because an error in last interval computing. Fixes #443.
commit 547b1d07 (2020-10-05): Bugfix: interval stats announced incorrectly because an error in last interval computing. Fixes #443.