OpenTracksApp/OpenTracks #446: Export/Import GPX without heart rate data end up with heart rate data.

rgmf, 2020-10-06:
**Describe the bug**
I've exported and imported all my OpenTracks tracks and some of them have ended up with heart rate datas that I can see in graphs tabs.

I've never used heart rate sensor and in GPS files there are not any heart rate data.

I have some gpx files I know they result in this weird situation, so I'm going to try to catch what happening

An image where you can see heart rate data but I don't know what they come from:
[image]

**Technical information**
 - Device: Pixel 2XL
 - OS: LineageOS 17.1
 - Version: 3.11.0

dennisguse, 2020-10-06:
Quick guess: the shown heartrate is the speed (or rather the inverse of speed aka pace).
Could that be the case?

rgmf, 2020-10-06:
> Quick guess: the shown heartrate is the speed (or rather the inverse of speed aka pace).
> Could that be the case?

Yes, I was thinking the same that you because values are really small, just like speed values in the gpx file.

closed 2020-10-08 by rgmf

rgmf, 2020-10-08:
@dennisguse very easy... a couple of breaks in a switch/case statement was missing. Fixed and merge onto main.