OpenTracksApp/OpenTracks #447: Since Bluetooth discovery dialog was removed, my HR sensor is no longer lister

vanous, 2020-10-06:
Thank you for all your great work!

The BT discovery dialog was removed some time ago (about a year ago). As i had my HR sensor (00002a37... GATT characteristics supported device) already set in OT, all was working even after that change. Unfortunately, after uninstall and new installation, newly installed OT does not allow me to select my device, in the Settings → Bluetooth → Heart rate → i have " * None ".

I can confirm, however, with nRF toolbox:

[image]

that my sensor works. The sensor is Amazfit Bip with HR access enabled for 3rd party software (making the sensor available as generic GATT device). So let me repeat - this device used to work normally with OT, but i have no longer change to actually select it to be able to use it.

Maybe i have missed something? Thank you very much.

dennisguse, 2020-10-06:
If I got it right, the discovery for this sensor is not working.
This code was no change since almost a year and it is working fine for me.
So, I guess it is either an issue with the sensor or the phone.
Could it be that the sensor requires to be paired or is it already paired?

Due to the lack of a sensor, I cannot test anything...

vanous, 2020-10-06:
Thank you.

>This code was no change since almost a year and it is working fine for me

I think that is the change i am referring to :) I was simply too busy to make a report back then. Plus, i was hoping somneone else might report this.

>Could it be that the sensor requires to be paired or is it already paired?

The sensor is paired. It is a regular, standard  00002a37... GAT characteristics based sensor: https://duckduckgo.com/?t=ffsb&q=00002a37+GATT

dennisguse, 2020-10-06:
What happens if you "unpair" the sensor?

My issue is that I have a simple sensor (not a watch) and discovery works nicely.
I am not even sure that my sensor can be paired...

vanous, 2020-10-06:
This is what BLExplorer shows:

[image]

This is the same in text form:

characteristic UUID: 00002a37-0000-1000-8000-00805f9b34fb
service UUID: 0000180d-0000-1000-8000-00805f9b34fb
value: 4b = 19200 = K 

Let me try to unpair...

vanous, 2020-10-07:
OK, removing the filter from the BLE scanner:

`ScanFilter.Builder scanFilterBuilder = new ScanFilter.Builder()`;

or this way 

`//scanFilter.add(scanFilterBuilder.build());`

Makes the discovery to work. Of course, you most likely get other devices into the list, which do not belong to this GATT Service. After selecting my device, HR starts working right away and it's name is also correctly used on the Recording screen. I mention this, because in the Preferences → Bluetooth → Heart Rate→ it is listed with MAC address, although the popup dialog shows the correct device name. (this is probably another issue).

Here is the reason: https://github.com/iDevicesInc/SweetBlue/wiki/Android-BLE-Issues
Bottom line from all advice i could gather: scan broadly and use homegrown filtering.

Originally prepared comment with various links, before i found the "Android BLE Issues" list.

---
for the record only:
Based on this, i was searching and found this, mentioning the same issue: https://stackoverflow.com/questions/45622605/bluetooth-scanfilter-uuid-in-android-7-1-1-not-working-but-works-in-6-0-1

So i tried searching more and found this: https://stackoverflow.com/questions/29664316/bluetooth-le-scan-filter-not-working , but i am not sure if the explanation really applies. 

There are more...

https://stackoverflow.com/questions/35072596/how-does-scanfilter-based-on-mac-addresses-work
https://stackoverflow.com/questions/34065210/android-ble-device-scan-with-filter-is-not-working

dennisguse, 2020-10-07:
@vanous What OS version do you use?

vanous, 2020-10-07:
Tested on this on Samsung Android O (8) and i also tested on LOS 17.1 (Android 10). Exactly same results.

Removing the scanFilter just makes things work :)

vanous, 2020-10-07:
I have never done any BLE related coding... 
but maybe this is the reason?
result.getScanRecord().getServiceUuids() 
returns 
[0000fee0-0000-1000-8000-00805f9b34fb]

WotblitzReversing, 2020-10-08:
Hey Vanous, Where could i contact you, it's about Dava Engine
Need your help dude, i knew you long time ago
Give me your Discord account if you still have it
Thanks you

dennisguse, 2020-10-11:
@vanous Could we scan for all and then filter the result based upon the provided services manually?

vanous, 2020-10-11:
> @vanous Could we scan for all and then filter the result based upon the provided services manually?

that would be probably the best, bbut when i tired this, i could not get all services listed, meaning the filter didn't work for me. But what i did on my side was that if i disable the filter completely, i can choose a particular device and it works.
The only problem this has is that the user must know what device provides what features/sensors...

davidoe, 2020-10-22:
My "Polar H7" cannot be found, too.

dennisguse, 2020-10-26:
@davidoe Can you check that your device is working with another app?
Because I had one of those and tested with it.
The battery might be empty.

vanous, 2020-10-28:
@dennisguse what do you see as a good solution? For me, creating a filter didn't work, but admittedly i didn't try very hard, because of the layer this must be built at. One possibility i could see is to provide filtered view (as you do now) and the possibility to turn the filter off, which raises complexity (another switch). Or, in the list of found devices, have as the last item ~ "Scan for all types of devices around", which would perform unfiltered scan.

dennisguse, 2020-10-31:
@vanous Don't know - none of the options is really cool.
I don't really want to invest time into a workaround, because Android internally is doing something weird.
At the moment, I will invest my time into refactoring the internal data structure (i.e., #192; we need this for import/export and synchronization) - we have some nasty problems there that need to be solved first.

If you have some spare time and can figure out a way to implement it - feel free to.

dennisguse, 2020-11-08:
@vanous Could it be that Android is caching the BLE services provided by the device?
Can you try to activate the heart rate and check if another Android device can connect to it directly?

See https://stackoverflow.com/questions/50739085/how-to-refresh-services-clear-cache

vanous, 2020-11-29:
@dennisguse sorry for late reply :) this doesn't seem to be the problem. Yes, another apps can connect and it works, today i tested nRF toolbox and also FitoTrack. I really would suggest to disable the filter and show all BLE devices around. The list will be much longer, but whether this is Android issue or BT/BLE stack issue, at this point, end users suffer and cannot use their devices. I can see that both nRF toolbox and FitoTrack have chosen this.

dennisguse, 2020-12-01:
@vanous I ordered a MiBand 3, so that I can some testing.
Stay tuned ;)

vanous, 2020-12-02:
> @vanous I ordered a MiBand 3, so that I can some testing.
> Stay tuned ;)

Awesome!

If you go the Gadgetbridge route, feel free to ask me anything ;)

dennisguse, 2020-12-03:
I got the Miband3.
Quick test: with nRF I only got HRM if device was paired (then it could be selected) and but only got an actual value/sensor started when triggered on the Miband3. HRM sensor not found via Android BLE scan (OpenTracks, RunnerUp, nRF).

vanous, 2020-12-03:
hahah, not ehough HRM samples, programmers problem :) let me explain, i had the same issue :)

The HRM are only sent if they are being done (makes sense). Multiple ways to trigger them:
1) look at the "infoL screen. Not sure MB3 has it, newer bands do. This is not important, because it still is only one measurement.
2) start an activity (on the band, select exercise or treadmill)
3) enable continuous HR measurement. this does **two** things: - takes a measurement "at least as often as set", for example every hour, but **at the same** time, if you start moving (doesn't happen behind keyboard :) ) this activity is detected and HRM are being taken to track the activity. So in fact, there is no need to "start an activity" to take HRM, simply use the continuous measurements. And if you have "make HRM visible via BLE" enabled, you get a nice HRM sensor.

vanous, 2020-12-04:
Btw, just related to pairing:

Using BLE-Monitor (available via Izzy's repo), i can get HRM from Bip even **without pairing**. Same setup but with MB3 did not provide HRM for me. MB3 is newer then the Amazfit Bip... ¯\_(ツ)_/¯.

dennisguse, 2020-12-04:
Got it! The Miband3 does not announce the heart rate support in it's BLE announcements packages and thus ScanFilter ignores it.
I added the Miband3's service UUID that it announces to the `ScanFilter` and then it is discovered correctly.

Although the setup is hard:
1. Miband: factory reset
2. Gadgetbridge: pair
3. OpenTracks: enable 3rd-party HR sensor forwarding
4. Gadgetbridge: disconnect
5. OpenTracks setings: select heart rate sensor
6. Gadgetbridge: connect
7. Miband: start  "treadmill" (triggers continuous heart rate measurement)
8. OpenTracks: start activity

NOTE: if "treadmill" is not started, triggering individual heart rate measurements on the Miband3 works as well (reports only this measurement)

Pairing via Android's Bluetooth Setting does not seem to be required.

@vanous Can you test this?
And can you create PRs for other devices (adding there UUID)?
I used `bluetoothctl` with `scan on` (on a notebook with Debian) and just waited for my disconnected Miband3 to send BLE announcements (I got the BLE address from Gadgetbridge).

Sidenote: I will stick with my OH1 - way easier to setup :partying_face:

vanous, 2020-12-05:
1) The setup sounds complicated, but if you have MiBand devices already in Gadgetbridge, there is pretty much no setup. 1) they are paired, 2) they might have HR exposed, if not, exposing it is a simple preference setting. Disabling it can be useful to preserve a bit of battery. 3) to have HR samples, the periodic HR measurement (1 per hour) is sufficient, because the band itself detects activity and provides full sampling of the activity, while one also has whole day sampling.

2) I can see devices in OT and i tested if they work:

[image]

I can confirm that if the device is/**was** paired, i can get HRM. **_Maybe_** even unpaired devices can do this but for me this didn't work. The problem (maybe only with Huami devices) is that if you pair and then unpair, the device still is paired, while the phone is not. This relates to your years old question in Gb issue tracker. But i tested this with multiple bands, so this feels confirmed: bands that have never been paired do not provide HRM, bands which are paired or were paired/unpaired do provide HRM.

There is a setting in the Zepp/MiFit app to allow visibility of the device itself. This is independent of the HRM exposure settings. If this "device is visible" was not enabled, i could not see the device in OT. So your filter works correctly, **_i think_**. We do not have the "device visible" setting in Gadgetbridge.

3) Your solution works. And you yourself might be wondering how come i can see some many device above, since only one UUID is in the list. Seems all Huami use the same UUID (see the UUID listing below).

I tried `bluetoothctl` but only once i managed to get the listing of the UUIDs, i think you mean this:

`[CHG] Controller de:ad:be:af:7B:89 UUIDs: 00001800-0000-1000-8000-00805f9b34fb`

plus many other lines, probably from other devices.

What i however see in info <device> is this:

`UUID: Anhui Huami Information.. (0000fee0-0000-1000-8000-00805f9b34fb)`
and also this
`ServiceData Key: 0000fee0-0000-1000-8000-00805f9b34fb`

So i have some reliable way to check. Good.

So i checked two bands: one with HRM exposed, one without.

The above (UUID and ServiceData Key) is the same for my MB3 and also for Amazfit Bip, therefore the conclusion that the UUID is probably the same.

For devices that have HR exposed, the `ServiceData Key: 0000fee0-0000-1000-8000-00805f9b34fb` is present, but if the HR is not exposed, there is no service data key. This actually means, that the list in OT (screenshot above) shows all devices with UUID of `0000fee0-0000-1000-8000-00805f9b34fb` no matter if they expose the HRM or not. This got me thinking that your solution is not good, because it is similar to my proposal (to just open up the filter)... but..:

So then, i tried both OT and nRF Toolbox and tried to get HRM from the band which doesn't `expose HRM to 3rd parties`. Hold and behold, it still does provide HRM, it is just not announcing it via the `ServiceData Key: 0000fee0-0000-1000-8000-00805f9b34fb`.

Summary attempt:

- `ServiceData Key: 0000fee0-0000-1000-8000-00805f9b34fb` is independent of the fact that Huami devices provide HRM
- if device was not paired but it is made visible and has HRM exposed, it is listed but doesn't provide data. I am not sure how other devices work/should work.

My level of understanding is still quite limited, but it seems like this is all and we can leave it as is, no further work is required and Huami devices are now supported in OT. :+1:

dennisguse, 2020-12-05:
Here is the output I used yesterday:
```
[CHG] Device C0:28:8D:DA:E8:B7 ManufacturerData Key: 0x0003
[CHG] Device C0:28:8D:DA:E8:B7 ManufacturerData Value:
  01 18 5b 00 60 09 00 00 00 00 00 00 22           ..[.`......."   
[CHG] Device C0:40:0F:85:26:F1 RSSI: -49
[CHG] Device C0:40:0F:85:26:F1 UUIDs: 0000fee0-0000-1000-8000-00805f9b34fb
[CHG] Device C0:28:8D:DA:E8:B7 ManufacturerData Key: 0x0003
[CHG] Device C0:28:8D:DA:E8:B7 ManufacturerData Value:
  01 18 5b 00 60 09 00 00 00 00 00 00 c0 28 8d da  ..[.`........(..
  e8 b7 22                                         .."             
[CHG] Device C0:28:8D:DA:E8:B7 ManufacturerData Key: 0x0003
```

`UUIDs` contains the announced _service uuids_ on BLE discovery requests.

We have to distinguish two things here.
1. When a Bluetooth devices ask around what devices are there (i.e., scanning), then devices receiving this request may respond and announce their present. To save energy the response is rather short. So, the Miband engineers decided to not send _all_ supported service UUIDs, but only their own (i.e., 0000fee0-0000-1000-8000-00805f9b34fb).
May respond: for example my Miband3 and also my OH1 do not reply if they are already (in this moment) connected to _any_ device. 
However, in this phase we actually cannot figure out if this device then supports/provide heart rate at all.
We just hope it does ;)
So, OpenTracks now accepts the Miband3 service UUID as a heart rate sensor supporting device.

2. When actually connecting to the device (to get heart rate sensor data), the _service discovery_ is conducted.
Here, one specific device is asked what service UUIDs it supports and it is expected that _all services_ are reported.
And in this case, the Miband3 _always_ responds with heart rate (i..e, `180D`) and then we try to get data from this service.
However, when the device is not setup correctly (not paired via Gadgetbridge) then the heart rate service can still be found, but it does not provide any data.

@vanous Totally forgot that I created this issue: https://github.com/Freeyourgadget/Gadgetbridge/issues/908
And I actually never used the Miband2 - just played around with it and then sold it again :)

If there are other devices that behave similarly like the Miband3, but announce a different UUIDs during discovery then we can just add them.

And can you add the devices you tested into the documentation (best case with a link to Gadgetbridge on how to actually setup)?
That would be great as it will help others to get avoid this (frustrating) complexity).
https://github.com/OpenTracksApp/OpenTracks/blob/main/README_TESTED_SENSORS.md

PS: If somebody ever discovers a nice documentation about Bluetooth/BLE, please let me know.

closed 2020-12-05 by dennisguse

vanous, 2020-12-18:
For any interested future reader, just for the record:

This has triggered another wave of curiosity, we investigated further and discovered that some bands can turn the BLE HR on/off while others have the BLE HR enabled all the time and it cannot be disabled. We have therefore [removed](https://codeberg.org/Freeyourgadget/Gadgetbridge/commit/4fb79ac2b256d1e7e904df01d7cbeb21ab28de0d) the BLE HR on/off settings from some bands supported by Gadgetbridge. I have updated [our wiki](https://codeberg.org/Freeyourgadget/Gadgetbridge/wiki/Huami-Heartrate-measurement#bluetooth-heart-rate-sensor) with slightly more details.

commit 76f59af5 (2020-12-05): BLE sensors: Miband3's is now considered a heart rate sensor during discovery.

Some BLE devices do not announce all there supported services in the BLE advertisement packages.
Android's ScanFilter decides based upon this information only.
When connecting to such a device (and trigger service discovery), the service can still be used.

Fixes #447.
PR #541 (2020-12-05): BLE sensors: Miband3's is now considered a heart rate sensor during discovery
Some BLE devices do not announce all there supported services in the BLE advertisement packages.
Android's ScanFilter decides based upon this information only.
When connecting to such a device (and trigger service discovery), the service can still be used.

Fixes #447.
commit 18bc100d (2020-12-05): BLE sensors: Miband3's is now considered a heart rate sensor during discovery.

Some BLE devices do not announce all there supported services in the BLE advertisement packages.
Android's ScanFilter decides based upon this information only.
When connecting to such a device (and trigger service discovery), the service can still be used.

Fixes #447.