OpenTracksApp/OpenTracks #448: Export in one file

petrhejna, 2020-10-06:
**Is your feature request related to a problem? Please describe.**
I like to do regular exports. But currently it will do export in multiple files.

**Describe the solution you'd like**
Single export in one file (archive).

**Describe alternatives you've considered**
To my knowledge, there is none.

**Additional context**
I use de-googlified phone. F-Droid only.

dennisguse, 2020-10-06:
Definitely possible; we even have code for this in place (for import/export KML/KMZ/GPX).
It is actually a leftover and not used (at the moment).

dennisguse, 2020-10-22:
@petrhejna just for clarification: why do you want one file instead of multiple ones?

petrhejna, 2020-10-22:
I would be handy to just upload one file to backup storage. Now I have to go through all files, select them and copy them. 

For example with Keybase app, you need to upload files even one by one.

pstorch, 2020-10-23:
While testing the current export function I recognized that, when using a cloud destination, hundreds of small files take longer to upload than one single big file. So I think this would be a good option.

dennisguse, 2020-10-23:
(For upload) exporting one file has the issue that destination provider needs to handle partial changes as otherwise the whole file needs to be completely uploaded again and again. And the file might grow big quite fast...

petrhejna, 2020-10-23:
@dennisguse That's true, I think having both option would be perfect.

maxolasersquad, 2020-10-31:
One thing that this would be helpful for is phone transfers. It is a lot more efficient to do a single export, transfer the single file, and then import that single file.

dapgo, 2021-01-23:
> 
> 
> @petrhejna just for clarification: why do you want one file instead of multiple ones?

@dennisguse exporting a single file ease the exchange and interoperability, and specially when filenames based in id.
As real examples of use,  you can send to a friend by different ways or just upload the file to a service as Strava.

dennisguse, 2021-01-23:
We need two things:
* a concept for the user interface (I assume we need to keep the export to individual files)
* and some actually tiny change to TrackWriter classes + TrackImporter classes for GPX/KML.

Coding is the easy part.
If someone has time to create a concept that would be great!

closed 2023-05-29 by dennisguse