OpenTracksApp/OpenTracks #451: Export all silently overwrites destination files

pstorch, 2020-10-07:
At the moment the "Export all" function exports all tracks over and over again. 
I would like to have a check if the file already exists at the destination being asked what conflict resolution should be performed:

Message: "This track already exists at the destination directory!"
Options:
- overwrite
- skip
- [ ] checkbox "do it for all conflicts"

As the track list in OT gets longer and longer a skip possibility for already exported tracks could significantly speed up the "Export all" process.
And if you have a cloud destination directory you don't cause unnecessary traffic.

rgmf, 2020-10-12:
@pstorch @dennisguse someone working on this? I have some time this week and I could take over this.

@dennisguse Should I wait until https://github.com/OpenTracksApp/OpenTracks/pull/454 merge?

dennisguse, 2020-10-12:
I guess, this would not conflict with #454.
Anyhow, it is ready, so if you could give it a spin, we can just finally merge #454 :)

commit 01f756c5 (2020-10-17): Export All: ask to the user when conflicts (track already exported in the selected directory) and show a summary. Fixes #451.
commit 4d1af344 (2020-10-17): Export All: ask to the user when conflicts (track already exported in the selected directory) and show a summary. Fixes #451.
commit f0dab3cc (2020-10-17): Export All: ask to the user when conflicts (track already exported in the selected directory) and show a summary. Fixes #451.
commit 440468e5 (2020-10-17): Export All: ask to the user when conflicts (track already exported in the selected directory) and show a summary. Fixes #451.
closed 2020-10-20 by rgmf