OpenTracksApp/OpenTracks #453: TrackDataHub for intervals

rgmf, 2020-10-09:
Intervals should use TrackDataHub to get data instead of Content Provider.

See here: https://github.com/OpenTracksApp/OpenTracks/pull/368#issuecomment-685105520

rgmf, 2020-10-10:
**Notes and thinking for now**

Speed/pace of intervals using TrackDataHub are wrong.

You see this issue only in long tracks, so I'm fairly sure that TrackDataHub filtering is not good.

Whay do I think that? Because comparing the intervals with others programs, they give me the same speed/pace like OT now but not like OT using TrackDataHub.

rgmf, 2020-10-10:
**Refactoring possibility:**

What about remove TrackDataHub?

Read this: https://github.com/OpenTracksApp/OpenTracks/issues/281#issuecomment-687651226

What about filtering?

Read this: https://github.com/OpenTracksApp/OpenTracks/issues/213

dennisguse, 2020-10-10:
Some (guessed) background infos:
The TrackHub implemention was implemented for really old Android devices (2012?).
It's job was to save resources for rendering on a map and diagram at one central point rather than in each component.

Nowadays performance should be a lesser constraint and the map was extracted to OSMDashboard.
So, it might be that we can remove this construct (don't know if there is other stuff that will break ><).
We would probably need to improve our diagram rendering (i.e., not render each trackpoint that is passed to the diagrams).

rgmf, 2020-10-10:
> Some (guessed) background infos:
> The TrackHub implemention was implemented for really old Android devices (2012?).
> It's job was to save resources for rendering on a map and diagram at one central [...]

So, I'm going to remove/rework all deprecated code in TrackDataHub... and I'll see :D

I'm thinking that the difference between record tracks and imported ones could get less. I mean this one: https://github.com/OpenTracksApp/OpenTracks/issues/275 could be improved so I let's get test it.

rgmf, 2020-10-10:
@dennisguse TrackDataHub works perfect... I didn't understand very well some things.

I think is good that TrackDataHub can limit the number of points to use because it's perfect for diagrams where you don't need all points to represente these diagrams. The only think I don't like is the interface: too much methods to implements, some of them empty in some parts. It's a bit ugly.

dennisguse, 2020-10-10:
I guess the data point filtering could be moved to the diagrams as it is not used by anybody else (anymore).
Then everybody else could use the contentprovider directly - so it might be possible to remove the TrackDataHub completely.

rgmf, 2020-10-10:
After deeping in TrackDataHub code I don't know if the best approach is using ContentProvider directly.

TrackDataHub has observers to Track, TrackPoint and Marker tables. When some of them change then it informs to all listeners. Also, when a track is recording it handles the track points feeding.

It becomes easier the code.

Practical example:
When the category of the recording track change (the user change the spinner icon) then Intervals, Chart and StatisticsRecording fragments need to do things. They know when this event trigger and the trach data because TrackDataHub inform them ("eh, track changed"). If we don't have TrackDataHub we need an observer in every part or an interface in the activity. I think the better apporach is the TrackDataHub where all this logic is.

The long story short: I'm not sure if we'll improve the code removing TrackDataHub.

commit 5e520193 (2020-10-10): Intervals: TrackDataHub instead of ContentProvider. Fixes #453.
commit 61cace28 (2020-10-11): Intervals: TrackDataHub instead of ContentProvider. Fixes #453.
commit 679c1d7e (2020-10-11): Intervals: TrackDataHub instead of ContentProvider. Fixes #453.
closed 2020-10-11 by rgmf