OpenTracksApp/OpenTracks #455: Accept donation in bitcoin :)

petrhejna, 2020-10-11:
Hi, great app I would love to donate some sats :)

dennisguse, 2020-10-12:
Nice!
@rgmf @pstorch 
Does anybody of you want to take about this or should I?

pstorch, 2020-10-12:
I'm not familiar with Bitcoin.

rgmf, 2020-10-12:
I'm not familiar with Bitcoin neither but I think is nice to add this possibility for people who prefers bitcoins.

petrhejna, 2020-10-12:
All around the world governments go brrrrrrrrr (printing money like crazy). It's good time to get into Bitcoin and Austrian economics. ;)

Bitcoin is the unconfiscable unprintable magic internet money. ;)

dennisguse, 2020-10-13:
Here we go: 
```
bitcoin:BC1QCKPLS3LPJLCPYXZM66HCY65HNESAS9ZGACY6YC
```

petrhejna, 2020-10-13:
Will you please make it PR and merge it to Readme (for other to be able to contribute as well).

Also QR code would be nice to have. :)

PR #458 (2020-10-13): Add option to donate via Bitcoin.
Fixes #455.
closed 2020-10-13 by dennisguse