OpenTracksApp/OpenTracks #457: GPS smoothning

bsbhadwal, 2020-10-13:
Background: I am a cyclist living in the Himalayas and elevation information is extremely important for the kind of riding I do. Use Strava and opentracks for recording.

Problem: there is a huge discrepancy in the elevation gained / loss when measuring the same route via Strava and opentracks. This in my knowledge is due to the fact that Strava employs some sort of GPS smoothing algorithm. However, this alao means that Strava elevation data is much closer to the ground truth. E.g same 32.16 km ride recorded on both platforms. Strava elevation gain - 630m (closer to the truth) opengps elevation gain 1316m (not accurate) 

Proposed solution: analysis of data post ride, after user stops recording. and removing any GPS outliers by smoothing the GPS profile and arriving at a more accurate elevation gain. This can be an experimental flag that users can turn on or off in the app settings. 

Happy to provide the GPS logs from both platforms for the same ride if needed.

dennisguse, 2020-10-13:
We are not using GPS at all for elevation.
At the moment, we only use the barometer.

Sensor fusion between GPS and barometer is relatively complicated.

Strava has some nice explanation how they do this: https://support.strava.com/hc/en-us/articles/216919447-Elevation-for-Your-Activity
They even use maps.

bsbhadwal, 2020-10-13:
So what about phones that do not have barometer? Mine does not. In such a case shouldn't GPS elevation data be used? If you agree, perhaps the ticket should be renamed to - support for elevation on non barometric devices

dennisguse, 2020-10-13:
Yes, we can also go for GPS only elevation gain!
A useful algorithm is described here: https://www.gpsvisualizer.com/tutorials/elevation_gain.html
But there will some pitfalls (e.g., we do not store all trackpoints) - so it is not straight forward and not possible in post-processing (at least not with the same result).

@bsbhadwal Just create a new issue (mark it as follow-up to this one) and close it.
Less clutter.

And I am slightly confused: how did you get an elevation gain in OpenTracks if you don't have a barometer (there should be no measurement happening)?
Or did you compute the elevation in another application afterwards (based upon GPS)?

bsbhadwal, 2020-10-13:
Hi you're right I imported the opentracks and strava data (using exported gpx files) in garmin basecamp and superimposed the elevation graphs. I will create the follow up issue. Thank you for your time!

hex-m, 2021-03-01:
> @bsbhadwal Just create a new issue (mark it as follow-up to this one) and close it.
> Less clutter.

The follow-up issue that was asked for: https://github.com/OpenTracksApp/OpenTracks/issues/619

closed 2021-04-06 by dennisguse