OpenTracksApp/OpenTracks #463: Use step count to compensate for bad GPS signal

christfried, 2020-10-18:
When I was using OpenTracks the other day my phone found its location pretty fast but then lost it immediately. It took around three kilometers till it found it again. So I lost around half of my trip just because of bad GPS. It was pretty cloudy that day but also in the woods this might be a problem for runners. (This might be my phone, since I have this error sometimes. Nonetheless it would be nice to have a backup plan ...)

Is there a way to compensate for bad GPS or even validate/correct the measured distance by using a step counter at the same time? Also to track at least something for the first few seconds/minutes until GPS signal is found?

dennisguse, 2020-10-18:
We ignore by default every GPS measurement that has a vertical accuracy of >50m (can be changed in the settings).
So, you might be able to finetune it to your phone by increasing the threshold.

About step sensor counting: 
OpenTracks stores it's data for each received GPS point - if there is none no data is saved so far.
We are limited to this internal data structure at the moment - and changing is a complicated operation.
So, I don't how we could store the step count if there is no (valid) GPS signal reported.
We even have trouble to export and import existing tracks with regard to restoring the speed / time of a track (See #275).

dennisguse, 2021-04-01:
All my above arguments are not valid anymore.
We could implement this - we would just need a test device.

christfried, 2021-04-01:
Wow I did not expect to hear something like this so soon. Great to hear!
A test device means that you would need a seperate phone that has a step counter and GPS? Are there still devices which lack that kind of functionality?

dennisguse, 2021-04-01:
I was actually thinking about separate sensors that are attached to the shoe and connected via Bluetooth.
https://fellrnr.com/wiki/Running_Sensors
https://www.bluetooth.com/specifications/specs/running-speed-and-cadence-profile-1-0/

Using the internal step counter of the phone would also be possible.
I actually did not think about this and I am not sure, if the accuracy will be high enough (as the phone can be anywhere on the body).

But both things are possible nowadays. is possible :)

@christfried I have a  phone available with TYPE_STEP_COUNTER.
i don't have any Bluetooth sensor.

christfried, 2021-04-02:
> I was actually thinking about separate sensors that are attached to the shoe and connected via Bluetooth.

This will definitely be the more accurate choice ... HOWEVER:

Most people using this app already got a step counter with them while tracking their acitivity and probably don't want to buy additional gear immediately. Even if the accuracy isn't the best, some data might be more accurate than no data at all (because of missing GPS). Does that make any sense ...?
Thank you for considering!

dennisguse, 2021-05-09:
@christfried thanks for the feedback.
I also checked with Trail-Sense https://github.com/kylecorry31/Trail-Sense (newest feature is step counting) and it works quite well.

Sadly, lack of time at the moment and too many open topics.
Adding BLE would be easier as we have already everything in place for this.

dennisguse, 2021-07-29:
We are adding support for BLE sensors for running.
Those estimate the distance properly on their own.

Superseded by #891.

closed 2021-07-29 by dennisguse

christfried, 2021-07-30:
Maybe I am not the right target group for this because I always want to use and carry as little technology as possible and as I see it: The more gadgets you take with you the more you can mess up in the way of not having charged everything, not having configured it the right way or in not getting / loosing bluetooth connection etc.
Still, thank you for considering.