OpenTracksApp/OpenTracks #464: Snap route into existing path to achieve more accurate distance measurements?

christfried, 2020-10-18:
My GPS tracks sometimes look a little bit shaky. I guess that this behaviour adds distance where it should not.

Would it help to snap the GPS track into an existing path, if there is one next to it? OsmAnd~ does this for Navigation. I wonder it the route would become more accurate that way?

pstorch, 2020-10-18:
OpenTracks itself cannot snap to an existing path. It has no map information. The OSMDashboard has a map but only offline maps have paths of the roads. I don't know if it is feasable to implement such a snap logic. But OSMDashboard already has something like a track smoothing, so GPS jitter is partly removed. You can set the smoothing threshold in meters.

christfried, 2020-10-18:
I just wondered if this would be possible to do because it could prevent measurement errors, right? Also all the live data (speed) depends on the accuracy of the distance measurement. I imagine implementing a whole map infrastructure might be a big undertaking, but maybe it would be worth it in the long run (pun intended). ;D

dennisguse, 2020-10-18:
No clue how to do this.
Some days ago we got a similar request for GPS elevation smoothing #457.
If we have map material (not only rendering), we could do this (as postprocessing?).
But it is a really huge effort...

dennisguse, 2021-07-29:
Nowadays, I would recommend using sensors (for cycling/ running) instead on relying on GPS.
Everything else is way too much effort (within OpenTracks) and data post-processing should take place in some other application.

closed 2021-07-29 by dennisguse