OpenTracksApp/OpenTracks #465: Accept GPX file via intent

vanous, 2020-10-18:
I am not sure if this is for Open Tracks or OSMDashboard. Probably OpenTracks could be better, to have the overview and possibility to use OSMDashboard, not sure. Let mw know, i can re-file under correct repo.

I log my tracks via my watch. This data is imported into Gadgetbridge. The produced GPX (contains subset of all collected data, such coordinates, elevation and and heart rate) can be exported via an intent. It would be really nice to use OT or OSMDashboard to view this data in addition to what Gb provides (we do not visualize GPX, but on the other hand, have plenty of other info).

I can of-course point OT to the directory of Gb, but all my tracks are being re-imported repeatedly, which takes a very long time and tracks are being duplicated.

Thank you very much

pstorch, 2020-10-18:
Ah, it is possible to transfer an issue between repositories. First time I did that :smile:

pstorch, 2020-10-18:
With the intent filters from OsmAnd https://github.com/osmandapp/OsmAnd/blob/2af22671a9fa05887f15bfd52b2d25bc777546ba/OsmAnd/AndroidManifest.xml#L158 I got OpenTracks to show up when sharing via FileManager, but still not via Locus Maps or OsmAnd.
I think I still don't understand these intent filters of Android correctly :disappointed:

dennisguse, 2020-10-18:
Back then I also experimented with the IntentFilters of OsmAnd, but with very little success.
It was really painful and I don't know why this thing is not working (anymore) :cry:

vanous, 2020-10-18:
We transfer uri to the file with gpx if not mistaken, this works well. Osmand, AAT and milkha open it up OK.

pstorch, 2020-10-19:
AAT has some other interesting intent filters for GPX files:
https://github.com/bailuk/AAT/blob/b1f2180e4d4be97e8c6f9e7e3c47e60fc20c524c/app/src/main/AndroidManifest.xml#L56

rgmf, 2020-10-19:
This works for me: https://github.com/OpenTracksApp/OpenTracks/pull/466

vanous, 2020-10-19:
Sounds good!

BTW, this reminds me our manifest :(

https://codeberg.org/Freeyourgadget/Gadgetbridge/src/branch/master/app/src/main/AndroidManifest.xml#L135

rgmf, 2020-10-19:
> Sounds good!
> 
> BTW, this reminds me our manifest :(
> 
> https://codeberg.org/Freeyourgadget/Gadgetbridge/src/branch/master/app/src/main/AndroidManifest.xml#L135

It's really ugly but seem the best option.

vanous, 2020-10-19:
Thank you all for looking into this. Very awesome to have this quick fixes, much appreciated! This will allow people with a smart band and Gadgetbridge (and other software) to push their activities to OT now much more easily.

I am still thinking if there is a way to utilize OT even more/differently, especially for those, who have some smart band without a gps. What i mean is not "intentionally captured activities", but activities detected via movement (steps, intensity) as captured by the watch/band... we have pushed initial concept this week: 

[image]

Right now, i feel that we are creating islands of data in our apps and have been thinking about building more bridges. Here, we have built/fixed another one. Very cool, thank you.

dennisguse, 2020-10-19:
@vanous That's a good idea - our (aka OpenTracks) only problem is that we cannot export "activities" without GPS locations.
Our data model resolves about these alone.
But you are right: we are creating data islands and that sucks in the long run as we cannot provide a consistent (eco system)-like experience, but all commercial apps will (they also do other stuff, but that is a different story).
So, in the long run, it would be awesome if we could import (automatically) everything into OpenTracks and that would be the one place to go.
Don't know how to continue from here - if we (as a community) need something like this, we need our stuff to get closer.

vanous, 2020-10-20:
> Don't know how to continue from here - if we (as a community) need something like this, we need our stuff to get closer.

@dennisguse at least it is good that we have a connection and know about each other now, so let's keep thinking about this. As for the export, we just started talking about it in our issue tracker.

closed 2020-10-21 by pstorch