OpenTracksApp/OpenTracks #468: Does work with Moofit ANT+ Heart Beat Sensor

Zugschlus, 2020-10-20:
Hi,

this is just a quick heads-up that I am using OpenTracks successfully with a "moofit ANT+ Herzfrequenzmesser" which can be ordered from Am*z*n. If you want me to include the link to the product, please let me know. This information can go into the tested sensors md file.

Greetings
Marc

dennisguse, 2020-10-20:
We only support BluetoothLE. Can you check with Bluetooth Explorer (in Fdroid) if you see this device?

Zugschlus, 2020-10-20:
The device also does BLE, that is just not mentioned in the product name. It shows up under "Bluetooth" in Opentracks as a Heart rate sensor with a MAC address. BLExplorer in Fdroid is 3 years old and does not seem to work on my Android 10 device.

dennisguse, 2020-10-20:
Can you check with another app like RunnerUp that the heart rate sensor is working?
if it can be discovered that means that the heart rate service could be discovered.
Then we might have some problem while connecting or decoding the data (rather unlikely - decoding is simple).
Or the device might not send data: heart rate sensors only send data if they detect a heart rate; it saves them energy.

Zugschlus, 2020-10-21:
I think we have a misunderstanding here. The heart rate sensor is working flawlessly with OpenTracks. I just wanted to give that information to you so that you can update your docs.

dennisguse, 2020-10-21:
@Zugschlus Got :)
Do you have a link; I just found amazon links...

Zugschlus, 2020-10-22:
I only have an Amazon link as well, https://www.amazon.de/gp/product/B08693K7TM, at least that's where I bought it.

commit 0bc9f6a0 (2020-10-26): Update README_TESTED_SENSORS.md

Fixes #468.
closed 2020-10-26 by dennisguse