OpenTracksApp/OpenTracks #470: KmzTrackImporter: Unable to parse kml in kmz

pstorch, 2020-10-21:
While testing #466 I stumbled upon the error that an exported kmz file from OT failt at import (or share from file manager) with "D/KmzTrackImporter: Unable to parse kml in kmz".
When I extract the doc.kml file from it, I can import it into OT without error.

rgmf, 2020-10-22:
I was trying to reproduce your error and I cannot to import all my OT exported tracks. This is the error:

```
java.lang.NullPointerException: Attempt to invoke virtual method 'int android.content.ClipData.getItemCount()' on a null object reference
```

Something was broken in PR #466 

@dennisguse don't prepare new release.

pstorch, 2020-10-22:
I'll check

pstorch, 2020-10-22:
I found the Nullpointer, yes was introduced by me.

@dennisguse can I remove the toString() hack? I get trouble in the debugger:

[image]

dennisguse, 2020-10-22:
@pstorch sure, it was only necessary for the refactoring.

pstorch, 2020-10-22:
Simple fix, I pushed it directly. Hope is it ok. @rgmf please give it a try.

closed 2020-10-22 by pstorch

rgmf, 2020-10-22:
@pstorch it works!!!

commit bbb91fed (2020-10-22): fix #470 Unable to parse kml in kmz