OpenTracksApp/OpenTracks #473: Optimize import with UUID check

pstorch, 2020-10-25:
While testing #472 I recognized a very slow import. Mainly because I reimported my exported tracks which I still have in OpenTracks and I have this "Prevent re-import of tracks" setting enabled. This causes each time the Unique constraint exception and then the cleanup. The cleanup deletes the imported track incl. a vacuum of the database which takes about 19 seconds each time on my device.
Wouldn't it be better to first check for the UUID and if it already exists in the database don't import the track so it doesn't need to be removed afterwards?

```
2020-10-24 23:36:07.316 4276-4786/de.dennisguse.opentracks.debug E/AbstractFileTrackImporter: Unable to import file
    android.database.sqlite.SQLiteConstraintException: UNIQUE constraint failed: tracks.uuid (code 2067 SQLITE_CONSTRAINT_UNIQUE[2067])
        at android.database.sqlite.SQLiteConnection.nativeExecuteForChangedRowCount(Native Method)
        at android.database.sqlite.SQLiteConnection.executeForChangedRowCount(SQLiteConnection.java:1074)
        at android.database.sqlite.SQLiteSession.executeForChangedRowCount(SQLiteSession.java:756)
        at android.database.sqlite.SQLiteStatement.executeUpdateDelete(SQLiteStatement.java:66)
        at android.database.sqlite.SQLiteDatabase.updateWithOnConflict(SQLiteDatabase.java:2220)
        at android.database.sqlite.SQLiteDatabase.update(SQLiteDatabase.java:2166)
        at de.dennisguse.opentracks.content.provider.CustomContentProvider.update(CustomContentProvider.java:281)
        at android.content.ContentProvider$Transport.update(ContentProvider.java:420)
        at android.content.ContentResolver.update(ContentResolver.java:2000)
        at de.dennisguse.opentracks.android.ContentResolverWrapper.update(ContentResolverWrapper.java:37)
        at de.dennisguse.opentracks.content.provider.ContentProviderUtils.updateTrack(ContentProviderUtils.java:305)
        at de.dennisguse.opentracks.io.file.importer.AbstractFileTrackImporter.onTrackEnd(AbstractFileTrackImporter.java:279)
        at de.dennisguse.opentracks.io.file.importer.KmlFileTrackImporter.endElement(KmlFileTrackImporter.java:121)
        at org.apache.harmony.xml.ExpatParser.endElement(ExpatParser.java:164)
        at org.apache.harmony.xml.ExpatParser.appendBytes(Native Method)
        at org.apache.harmony.xml.ExpatParser.parseFragment(ExpatParser.java:523)
        at org.apache.harmony.xml.ExpatParser.parseDocument(ExpatParser.java:484)
        at org.apache.harmony.xml.ExpatReader.parse(ExpatReader.java:322)
        at org.apache.harmony.xml.ExpatReader.parse(ExpatReader.java:285)
        at javax.xml.parsers.SAXParser.parse(SAXParser.java:390)
        at javax.xml.parsers.SAXParser.parse(SAXParser.java:187)
        at de.dennisguse.opentracks.io.file.importer.AbstractFileTrackImporter.importFile(AbstractFileTrackImporter.java:135)
        at de.dennisguse.opentracks.io.file.importer.KmlFileTrackImporter.importFile(KmlFileTrackImporter.java:38)
        at de.dennisguse.opentracks.io.file.importer.KmzTrackImporter.parseKml(KmzTrackImporter.java:260)
        at de.dennisguse.opentracks.io.file.importer.KmzTrackImporter.findAndParseKmlFile(KmzTrackImporter.java:188)
        at de.dennisguse.opentracks.io.file.importer.KmzTrackImporter.importFile(KmzTrackImporter.java:83)
        at de.dennisguse.opentracks.io.file.importer.ImportService.importFile(ImportService.java:70)
        at de.dennisguse.opentracks.io.file.importer.ImportService.onHandleWork(ImportService.java:50)
        at androidx.core.app.JobIntentService$CommandProcessor.doInBackground(JobIntentService.java:392)
        at androidx.core.app.JobIntentService$CommandProcessor.doInBackground(JobIntentService.java:383)
        at android.os.AsyncTask$3.call(AsyncTask.java:378)
        at java.util.concurrent.FutureTask.run(FutureTask.java:266)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
        at java.lang.Thread.run(Thread.java:919)
2020-10-24 23:36:07.316 4276-4786/de.dennisguse.opentracks.debug W/CustomContentProvider: Deleting table trackpoints
2020-10-24 23:36:07.325 4276-4786/de.dennisguse.opentracks.debug W/CustomContentProvider: Deleting table waypoints
2020-10-24 23:36:07.327 4276-4786/de.dennisguse.opentracks.debug W/CustomContentProvider: Deleting table tracks
2020-10-24 23:36:07.332 4276-4786/de.dennisguse.opentracks.debug I/CustomContentProvider: Vacuuming the database.
```

The slow vacuum of the database is also executed when deleting a few tracks from the TracksList, then each track is deleted separately with a vacuum call. 
I think we need to refactor the delete method which includes the vacuum db call: https://github.com/OpenTracksApp/OpenTracks/blob/bbb91fedc50c44e27de2eb13a7cd7dd065119190/src/main/java/de/dennisguse/opentracks/content/provider/CustomContentProvider.java#L92

rgmf, 2020-10-25:
@pstorch, @dennisguse, can we delete "Prevent re-import of tracks" adding a question with a checkbox in ImportActivity every time a track already exists? Like the one in ExportActivity with override/skip buttons when a file already exists.

Look 281 - 288 lines at AbstractFileTrackImporter.java
```java
} catch (SQLiteConstraintException e) {
    if (PreferencesUtils.getPreventReimportTracks(context)) {
        throw e;
    }

    //TODO This is a workaround until we have proper UI.
    trackData.track.setUuid(UUID.randomUUID());
    contentProviderUtils.updateTrack(trackData.track);
}
```
@pstorch I don't know if it's related with this issue but I think is here where the Exception is triggered.

dennisguse, 2020-10-25:
Exactly, that is the place. For me, the checkbox approach would be great!

pstorch, 2020-10-25:
Asking the user instead of the Prevent re-import of tracks" would be fine.

The SAX parser really adds some complexity here. I guess the reason was performance and less memory consumption. But in this case it results in data partially written to the database. 

Maybe in the onEndElement() method when recognizing the TAG_UUID we could check if it already exists in the database and cancel the track reading (or skip until next track).

Also the transaction handling could be improved.

The deleteTrack method calls three times the delete method which creates a transaction by itself:

https://github.com/OpenTracksApp/OpenTracks/blob/bbb91fedc50c44e27de2eb13a7cd7dd065119190/src/main/java/de/dennisguse/opentracks/content/provider/ContentProviderUtils.java#L192

https://github.com/OpenTracksApp/OpenTracks/blob/bbb91fedc50c44e27de2eb13a7cd7dd065119190/src/main/java/de/dennisguse/opentracks/content/provider/CustomContentProvider.java#L113

dennisguse, 2020-10-25:
If we delete several tracks (during delete or import), we only need to vacuum after the last one.
The Sqlite's vacuum operation basically rewrites the whole database into a new file replacing the previous one.
Another option: we leave it to the database and use `auto_vacuum=FULL`; then it vacuums after _every_ transaction. Don't know if this is a good idea

And another thing: after DB upgrades that lead to re-creation/deletion of tables, we should also run vacuum (we didn't so far).

https://www.sqlite.org/lang_vacuum.html

pstorch, 2020-10-25:
The automatic vacuum doesn't sound good to be honest. I think the best way is to have the vacuum as a separate call and give the responsibility to the caller to call it after all deletions are done.

dennisguse, 2020-10-26:
#317 is for DB optimization.
I will add the DB vacuum optimization there as well.

Some other (potential) optimizations:
1. Parse the KML/GPX twice: parse first for UUID only and if not found/found but not know, do the actual import by parsing the file again.
2. Use ONE db transaction; when encountering that the UUID is already in the DB, rollback the transaction.

pstorch, 2020-11-06:
I was looking into it a little bit:

I tried checking if the UUID exists right when the `opentracks:trackid` element is read. Then cancel the reading by throwing the ImportAlreadyExistsException before reading all the trackpoints. This doesn't help because the track is already partially inserted into the DB at `onTrackStart()` and needs to be deleted which causes a vacuum. 

Option 1 (as proposed by @dennisguse): read the KML/GPX twice. Would need a separate SAXParser for each file type or a special mode where only the UUID is read. But I fear code complexity here. Also don't know the performance impact, if all files are read twice.

Option 2  (as proposed by @dennisguse): would mean the Importers need to know about and handle the DB transaction. Start transaction when start reading the file, commit transaction when file reading is finished. I somehow don't feel well with this. 

Option 3: remove the vacuum from the delete method and do it separately after import completed with errors. But then also the normal delete calls need to handle the vacuum separately. Haven't analyzed the impact yet.

Option 4: don't partially write the track into the DB. Keep it in memory until the track is fully read, then the deletion would not be necessary on UUID conflict. I don't know how big the tracks can get and if we could cause an OOM here. On the other hand there is already the MAX_BUFFERED_LOCATIONS which holds 512 locations in memory before flushing to DB. I also read the Track with all Trackpoints into memory in the OSMDashboard (for track smoothing).

At the moment I think Option 3 is the one with the least risk and impact. But I don't have the full picture yet.
Option 4 would result in the cleanest code, but has some risk of OOM.
Any additional thoughts or opinions?

rgmf, 2020-11-07:
I would like to add an idea around Option 2 (two step import): we can load in memory all `<opentracks:trackid>` tag's values into memory with something like XPath and before importing look for the trackid in memory. If there is there then throw ImportAlreadyExistsException, otherwise do the import.

In that way it's not necessary the separate SAXParser.

pstorch, 2020-11-07:
Isn't that also reading the full file twice? Otherwise how can an XML parser find all these tags. Does the SAX parser support XPath?

rgmf, 2020-11-07:
Yes, the file would read twice. I only point XPath out in the case option 2 would be the selected option. Only because the code would be simpler than using SAX.

dennisguse, 2020-11-07:
I guess the SAXParser would be very short as we only have to react to one element and wait for it's content.
Something like 20 lines or so, but I hope that XPath also would behave very similar to SAX as we only have one element that we are looking for.

pstorch, 2020-11-07:
Yes, XPath could be an option to an SAX handler for only the UUID.

Other consideration: when we (in the future) want to support multiple tracks in one file, will that approach still work? The we would need a bit more context, not only a list of uuids read in a first run.

dennisguse, 2020-11-07:
@pstorch Valid point -> I have a draft to enable multi-track import (not ready I need to do it again) as it the single track import has a lot of unneeded complexity (i.e., `ContentProviderUtils.cleanTrack()`).

pstorch, 2020-11-07:
My inner child would just like to do Option 4, as it would be the easiest / cleanest and ignore the OOM risk. Don't know if it really is one.

dennisguse, 2020-11-07:
And option 5: we may use a long running database transaction.
So, far we insert _some_ data and immediately commit the transaction (per track as well as for MAX_BUFFERED_LOCATIONS trackpoints and markers).
So, we end up with a lot of transactions (also for the rollback).
What about writing to the DB directly from the import code (not using the content provider) and starting a transaction on startTrack and commit it on trackEnd?
If the UUID is already there the whole transaction will fail and no data was changed.
This will also work for multi-track files.

pstorch, 2020-11-07:
Isn't it Option 2 already?

dennisguse, 2020-11-07:
Updated to Option 2.1 - for option 2 I assumed that we would use the contentProvider.

rgmf, 2020-11-09:
I've taken this... if nobody is working on this I'm going to look into this altough it seems difficult (we'll see).

commit 413ec8d4 (2020-11-13): Import files by track in memory (avoid clean track when error). Fixes #473.
commit 09245fba (2020-11-23): Import files by track in memory (avoid clean track when error). Fixes #473.
commit 9c9bdde6 (2020-11-25): Import files by track in memory (avoid clean track when error). Fixes #473.
closed 2020-11-25 by rgmf