OpenTracksApp/OpenTracks #482: While recording handle GPS disabled

dennisguse, 2020-10-30:
While recording a user might disable the GPS in Android.
What should we do then?

At the moment, we just continue recording, but we don't get any GPS coordinates delivered.

rgmf, 2020-12-31:
I like how OT handles this situation: it shows a notification on the notification bar and stop moving time (total time go on).

But it's also truth that user can overlook the notification bar and don't realize that GPS was disabled. I'd add an SnackBar or something like this.

dennisguse, 2022-01-25:
@rgmf Would you have the time to build some GPS feedback for the StatisticsRecordingFragment?
Some of the next steps is to make GPS optional and then it should be visible to user if it is on or not...
Technically, it is not an error anymore :)
(although we still request the permission  #1061)

rgmf, 2022-01-25:
@dennisguse okay, I'll give it a spin, so I'll see the best way to make GPS status visible to the user.

dennisguse, 2022-04-02:
@rgmf This was fixed right?

rgmf, 2022-04-02:
@dennisguse yes, it is!!! :tada:

closed 2022-04-02 by rgmf