OpenTracksApp/OpenTracks #483: While recording: show track statistics in notification

dennisguse, 2020-10-30:
Many smart watches can show notifications including the attached data.
As we have to show a notification anyhow, we can also present more data (like time, speed, distance, heartrate).

Would be especially interesting for running.

PS: I don't have a device for testing.

pstorch, 2020-10-31:
Yes, the Notification could show some more useful information. The GPS accuracy is not really interesting for me. I would like to see time, distance, pace, ...

dennisguse, 2020-11-01:
Sadly, it is not that simple to implement...
Everything is just mingled together in TrackRecordingService.
What we need to do react to `newTrackPoint()` and _after_ the TrackStatisticsUpdater was executed, update send this info to the NotificationManager.
And we need to make sure that we do something if the user is idle (i.e., update total time while not showing speed?).
But if we use a speed sensor (for cycling), we still should show the speed.

Long story short: somehow we need to integrate the data from TrackStatisticsUpdater and RemoteSensorManager together...

dennisguse, 2021-03-19:
This was actually implemented in #584.

closed 2021-03-19 by dennisguse