OpenTracksApp/OpenTracks #484: Auto-Export Tracks into Directory

boi4, 2020-10-30:
**Is your feature request related to a problem? Please describe.**

Hello OpenTracks-Team,
I'm a big fan of your software and use it on a regular basis to track my running. 
However, I would really like to synchronize my tracks from OpenTracks with my PC via Tools like Syncthing.

**Describe the solution you'd like**
An option to auto export new/updated tracks in a specified format into a specified directory.

**Describe alternatives you've considered**
* One could also sync the sqlite database with root access and then run the conversion process on a server. This is a bad solution as one would have to make the conversion code run on a different environment + one needs root access.
* Export each track per hand (a little bit cumbersome)

**Additional context**
-

Thank you!

pstorch, 2020-10-30:
Hello @boi4 this is already implemented in the current development version. I hope we can release it soon.

boi4, 2020-10-30:
Awesome. Thank you!

closed 2020-10-30 by boi4