OpenTracksApp/OpenTracks #486: Request for comments: internal data storage issues

dennisguse, 2020-10-31:
Affected issues: #375, #316, #275, #192, #301 (remotely).

# Problem

We cannot restore/recompute the TrackStatistics (mainly on import but also if we would remove TrackPoints from a track; see #375)
Fun fact: With regard to time, TrackStatistics are computed live while recording.

# Current situation
All our data EXPORT/IMPORT is based upon TrackPoints (either GPS-based or PAUSE/RESUME markers).
The later are also segment separator (either manually or automatically due to distance between two GPS-based TrackPoints).

A timestamp of a TrackPoint either use GPS time or device system time.
These timestamps cannot be compared to each other directly - worst case scenario: GPS week rollover.
The differences might also be smaller, but we cannot assume that the offset is constant.
Also the GPS hardware might apply some internal processing...

One brief example:
```
-> start recording
---> Segment 1
* TrackPoint/GPS: GPS_time=1;x=1/y=2
* TrackPoint/GPS: GPS_time=2;x=2/y=2
-> automatically create segment marker
* TrackPoint/PAUSE: DEVICE_time=3;x=100/y=0 --> contains special coordinates

---> Segment 2
* TrackPoint/GPS: GPS_time=4;x=4/y=2
* TrackPoint/GPS: GPS_time=5;x=5/y=2
-> pause recording
* TrackPoint/PAUSE: DEVICE_time=6;x=100/y=0 --> contains special coordinates

-> recording resumed
--> Segment 3
* TrackPoint/RESUME: DEVICE_time=7;x=200/y=0 --> contains special coordinates
...
```

On export, we just write the GPS-based TrackPoints; all device-based TrackPoints _are not_exported.
The device-based TrackPoints are used to create GPX/KML-segments (without exporting the DEVICE_time).

On import, we recreate device-based TrackPoints using the segments, but we _cannot restore the DEVICE_time (we actually use the current DEVICE_time on import).
And thus, we recomputing the TrackStatistics leads to incorrect results - basically everything that is affected by wrong segment duration (i.e., moving time, avg speed etc, overall track duration).
In fact, this also affects the starttime and the stoptime of a Track (also DEVICE_time).

Interesting: RESUME-TrackPoints are only written to DB, but actually never evaluated (basically just ignored on export).

# Idea: somehow we need to export/import TrackPoints that are based upon DEVICE_time.

## Export segment markers to denote segment start/end timestamps in exported files
One simple solution is to export device-based TrackPoints to export the timestamps of the segments correctly.
As device-based TrackPoints do not provide useful GPS coordinates, the GPS-coordinate of the previous/next TrackPoint could be used (depending if either segment start/segment end).
This can be done on export and reversed on import (if the first two coordinates are identical)

## GPS-time vs. device-time

To ensure that the time in exported files is monotonically increasing, we can use _either_:
1. DEVICE_time (can be added to every TrackPoint on creation), or
2. GPS_time.

_2nd option_: To be able to convert DEVICE_time to GPS_time, we would need to store device-time and GPS-time per TrackPoint - 
At least for the GPS_time TrackPoints before/after a DEVICE_time TrackPoint.

@pstorch @rgmf 
Did I miss something?
And what is your gut feeling?
I am the moment leaning towards the _2nd option_ as we do not modify the GPS_time if somebody needs it (for whatever use case).

Whatever, we do - it will be a really challenging operation :partying_face:

pstorch, 2020-10-31:
> As device-based TrackPoints do not provide useful GPS coordinates, the GPS-coordinate of the previous/next TrackPoint could be used (depending if either segment start/segment end).

That sounds good to me. As for pause we don't have another GPS update, we have to assume the user is still there. On resume could then use the first GPS update (next real Trackpoint) as it's coordinate.

> GPS-time vs. device-time

Why not use device-time all the time?

dennisguse, 2020-10-31:
@pstorch Time: don't ask me why, but it did not felt right.
But now that you are mentioning it: it totally makes sense - simplifies a lot of stuff!
And if we really need it, we can just re-add GPS-timestamps (for new tracks!).

rgmf, 2020-11-01:
> Why not use device-time all the time?
:+1: 

> In fact, this also affects the starttime and the stoptime of a Track (also DEVICE_time).

About starttime and stoptime, I think OpenTracks shouldn't start until get the first TrackPoint (GPS location). If user starts a record and stop it without points the track shouldn't be saved but discard.

In fact, if your GPS sensor is off OpenTracks can record an empty track with only total time (in the database we have a track but no any trackpoint). Then, if you export/import this empty track you will obtain a track with a total time of 0 seconds.

dennisguse, 2020-11-01:
@rgmf Makes total sense!
Some month ago, I was not a fan of this, but it nowadays I think we should go down this road!

dennisguse, 2020-11-07:
I guess this one blocks  #500.

dennisguse, 2021-02-13:
Changes are implemented and were released in v3.15.0; implemented in https://github.com/OpenTracksApp/OpenTracks/pull/553.

closed 2021-02-13 by dennisguse