OpenTracksApp/OpenTracks #491: Crash when trying to export all

maxolasersquad, 2020-11-01:
**Describe the bug**
When I try to export all of my tracks the application crashes.

**To Reproduce**
1. Tap the ellipses in the top-right
2. Tap settings
3. Tap "Export all"
4. Browse to the desired destination and then tap the "Use this folder button"
5. The application crashes.

**Technical information**
 - Pixel 3 XL
 - Android 11
 - OpenTracks v3.11.1-a42bd432e installed from F-Droid.

dennisguse, 2020-11-01:
Any chance to get a logcat?

rgmf, 2020-11-01:
I've had the oportunity of reproduce it in a Pixel 3a with Android 11 and last F-Droid OT's version. Here is what logcat says:

```
java.lang.SecurityException: Permission Denial: writing com.android.externalstorage.ExternalStorageProvider uri content://com.android.externalstorage.documents/tree/primary%3AOpenTracksBackup/document/primary%3AOpenTracksBackup from pid=26717, uid=10293 requires android.permission.MANAGE_DOCUMENTS, or grantUriPermission()
        at android.content.ContentProvider.enforceWritePermissionInner(ContentProvider.java:896)
        at com.android.externalstorage.ExternalStorageProvider.enforceWritePermissionInner(ExternalStorageProvider.java:154)
        at android.provider.DocumentsProvider.callUnchecked(DocumentsProvider.java:1119)
        at android.provider.DocumentsProvider.call(DocumentsProvider.java:1067)
        at com.android.externalstorage.ExternalStorageProvider.call(ExternalStorageProvider.java:647)
        at android.content.ContentProvider.call(ContentProvider.java:2448)
        at android.content.ContentProvider$Transport.call(ContentProvider.java:517)
        at android.content.ContentProviderNative.onTransact(ContentProviderNative.java:295)
        at android.os.Binder.execTransactInternal(Binder.java:1154)
        at android.os.Binder.execTransact(Binder.java:1123)
2020-11-01 17:46:39.032 26717-26963/? E/AndroidRuntime: FATAL EXCEPTION: Thread-2
    Process: de.dennisguse.opentracks, PID: 26717
    java.lang.NullPointerException: Attempt to invoke virtual method 'android.net.Uri androidx.documentfile.provider.DocumentFile.getUri()' on a null object reference
        at de.dennisguse.opentracks.fragments.ExportProgressDialogFragment$ExportThread.exportTrack(ExportProgressDialogFragment.java:219)
        at de.dennisguse.opentracks.fragments.ExportProgressDialogFragment$ExportThread.run(ExportProgressDialogFragment.java:200)
```

PS/ I also have the issue that was described in https://github.com/OpenTracksApp/OpenTracks/issues/439:

> Another minor issue is, when exporting a second time, the folder is shown, but the button at the bottom "use this folder for Export" is missing frowning_face

maxolasersquad, 2020-11-01:
I had to do some research on how to generate logcat logs.

```
11-01 12:17:22.728  3834  3900 E DatabaseUtils: Writing exception to parcel
11-01 12:17:22.728  3834  3900 E DatabaseUtils: java.lang.SecurityException: Permission Denial: reading com.android.externalstorage.ExternalStorageProvider uri content://com.android.externalstorage.documents/tree/primary%3AOpenTracks/document/primary%3AOpenTracks/children from pid=31012, uid=10457 requires that you obtain access using ACTION_OPEN_DOCUMENT or related APIs
11-01 12:17:22.728  3834  3900 E DatabaseUtils: 	at android.content.ContentProvider.enforceReadPermissionInner(ContentProvider.java:820)
11-01 12:17:22.728  3834  3900 E DatabaseUtils: 	at com.android.externalstorage.ExternalStorageProvider.enforceReadPermissionInner(ExternalStorageProvider.java:147)
11-01 12:17:22.728  3834  3900 E DatabaseUtils: 	at android.content.ContentProvider$Transport.enforceReadPermission(ContentProvider.java:684)
11-01 12:17:22.728  3834  3900 E DatabaseUtils: 	at android.content.ContentProvider$Transport.query(ContentProvider.java:239)
11-01 12:17:22.728  3834  3900 E DatabaseUtils: 	at android.content.ContentProviderNative.onTransact(ContentProviderNative.java:106)
11-01 12:17:22.728  3834  3900 E DatabaseUtils: 	at android.os.Binder.execTransactInternal(Binder.java:1154)
11-01 12:17:22.728  3834  3900 E DatabaseUtils: 	at android.os.Binder.execTransact(Binder.java:1123)
11-01 12:17:22.729 31012  4008 W DocumentFile: Failed query: java.lang.SecurityException: Permission Denial: reading com.android.externalstorage.ExternalStorageProvider uri content://com.android.externalstorage.documents/tree/primary%3AOpenTracks/document/primary%3AOpenTracks/children from pid=31012, uid=10457 requires that you obtain access using ACTION_OPEN_DOCUMENT or related APIs
11-01 12:17:22.730  3834  3900 E DatabaseUtils: Writing exception to parcel
11-01 12:17:22.730  3834  3900 E DatabaseUtils: java.lang.SecurityException: Permission Denial: writing com.android.externalstorage.ExternalStorageProvider uri content://com.android.externalstorage.documents/tree/primary%3AOpenTracks/document/primary%3AOpenTracks from pid=31012, uid=10457 requires android.permission.MANAGE_DOCUMENTS, or grantUriPermission()
11-01 12:17:22.730  3834  3900 E DatabaseUtils: 	at android.content.ContentProvider.enforceWritePermissionInner(ContentProvider.java:896)
11-01 12:17:22.730  3834  3900 E DatabaseUtils: 	at com.android.externalstorage.ExternalStorageProvider.enforceWritePermissionInner(ExternalStorageProvider.java:154)
11-01 12:17:22.730  3834  3900 E DatabaseUtils: 	at android.provider.DocumentsProvider.callUnchecked(DocumentsProvider.java:1119)
11-01 12:17:22.730  3834  3900 E DatabaseUtils: 	at android.provider.DocumentsProvider.call(DocumentsProvider.java:1067)
11-01 12:17:22.730  3834  3900 E DatabaseUtils: 	at com.android.externalstorage.ExternalStorageProvider.call(ExternalStorageProvider.java:647)
11-01 12:17:22.730  3834  3900 E DatabaseUtils: 	at android.content.ContentProvider.call(ContentProvider.java:2448)
11-01 12:17:22.730  3834  3900 E DatabaseUtils: 	at android.content.ContentProvider$Transport.call(ContentProvider.java:517)
11-01 12:17:22.730  3834  3900 E DatabaseUtils: 	at android.content.ContentProviderNative.onTransact(ContentProviderNative.java:295)
11-01 12:17:22.730  3834  3900 E DatabaseUtils: 	at android.os.Binder.execTransactInternal(Binder.java:1154)
11-01 12:17:22.730  3834  3900 E DatabaseUtils: 	at android.os.Binder.execTransact(Binder.java:1123)
11-01 12:17:22.731 31012  4008 W DocumentsContract: Failed to create document
11-01 12:17:22.731 31012  4008 W DocumentsContract: java.lang.SecurityException: Permission Denial: writing com.android.externalstorage.ExternalStorageProvider uri content://com.android.externalstorage.documents/tree/primary%3AOpenTracks/document/primary%3AOpenTracks from pid=31012, uid=10457 requires android.permission.MANAGE_DOCUMENTS, or grantUriPermission()
11-01 12:17:22.731 31012  4008 W DocumentsContract: 	at android.os.Parcel.createExceptionOrNull(Parcel.java:2373)
11-01 12:17:22.731 31012  4008 W DocumentsContract: 	at android.os.Parcel.createException(Parcel.java:2357)
11-01 12:17:22.731 31012  4008 W DocumentsContract: 	at android.os.Parcel.readException(Parcel.java:2340)
11-01 12:17:22.731 31012  4008 W DocumentsContract: 	at android.database.DatabaseUtils.readExceptionFromParcel(DatabaseUtils.java:190)
11-01 12:17:22.731 31012  4008 W DocumentsContract: 	at android.database.DatabaseUtils.readExceptionFromParcel(DatabaseUtils.java:142)
11-01 12:17:22.731 31012  4008 W DocumentsContract: 	at android.content.ContentProviderProxy.call(ContentProviderNative.java:732)
11-01 12:17:22.731 31012  4008 W DocumentsContract: 	at android.content.ContentResolver.call(ContentResolver.java:2405)
11-01 12:17:22.731 31012  4008 W DocumentsContract: 	at android.provider.DocumentsContract.createDocument(DocumentsContract.java:1371)
11-01 12:17:22.731 31012  4008 W DocumentsContract: 	at androidx.documentfile.provider.TreeDocumentFile.createFile(TreeDocumentFile.java:53)
11-01 12:17:22.731 31012  4008 W DocumentsContract: 	at androidx.documentfile.provider.TreeDocumentFile.createFile(TreeDocumentFile.java:45)
11-01 12:17:22.731 31012  4008 W DocumentsContract: 	at de.dennisguse.opentracks.io.file.ExportFile.getExportDocumentFile(ExportFile.java:13)
[... cut ...]
11-01 12:17:22.867  1465  1521 W WindowManager: 	at android.os.Handler.handleCallback(Handler.java:938)
11-01 12:17:22.867  1465  1521 W WindowManager: 	at android.os.Handler.dispatchMessage(Handler.java:99)
11-01 12:17:22.867  1465  1521 W WindowManager: 	at android.os.Looper.loop(Looper.java:223)
11-01 12:17:22.867  1465  1521 W WindowManager: 	at android.os.HandlerThread.run(HandlerThread.java:67)
11-01 12:17:22.867  1465  1521 W WindowManager: 	at com.android.server.ServiceThread.run(ServiceThread.java:44)
```

maxolasersquad, 2020-11-01:
This does appear to be a duplicate of https://github.com/OpenTracksApp/OpenTracks/issues/439 which is marked as Closed (which I guess is why I didn't find it when searching).

maxolasersquad, 2020-11-01:
It looks like #439 was attempted to be fixed in https://github.com/OpenTracksApp/OpenTracks/commit/bc04efbaac52772f13764b250b4d3957d909dd25 which was just before 3.11.0 was released, and I'm on 3.11.1. Am I seeing a slightly different bug, or maybe the fix in that commit didn't fully resolve it?

dennisguse, 2020-11-01:
Looks like we miss a permission.
Can try to add android.permission.MANAGE_DOCUMENTS to the manifest and see if this solves the issue?

rgmf, 2020-11-01:
Afte reading the new changes in Android 11, I think we need doing this: https://developer.android.com/preview/privacy/storage#all-files-access

More info: https://medium.com/androiddevelopers/android-11-storage-faq-78cefea52b7c

PS/ With MANAGE_DOCUMENTS in the manifest doesn't work for me.

pstorch, 2020-11-01:
I think I found something. Will send a PR for testing.

maxolasersquad, 2020-11-02:
@pstorch If you can get me an apk I'll be happy to install it and see if it works.

pstorch, 2020-11-02:
@maxolasersquad this is a debug build with a possible fix:
[OpenTracks-debug.zip](https://github.com/OpenTracksApp/OpenTracks/files/5477015/OpenTracks-debug.zip)

Can be installed in parallel to the release version.

dennisguse, 2020-11-03:
I would love to move this on into the upcoming release - we are slightly overdue ;)
Any news on this one?

pstorch, 2020-11-03:
The fix worked for @rgmf and for me in the emulator. I don't know if @maxolasersquad had time to do a third test on his physical device.

closed 2020-11-03 by dennisguse

maxolasersquad, 2020-11-05:
I finally got to test it this evening. It worked perfectly.

commit 90a8fc89 (2020-11-03): fix #491 crash export on android 11