OpenTracksApp/OpenTracks #492: `SearchListActivity` and `SearchEngine` need love

dennisguse, 2020-11-01:
I guess, I read `SearchListActivity` and `SearchEngine` for the first time...

Issues:
* everything is loaded from database
* data is kept in memory
* data is unmarshalled, marshalled, and then unmarshalled again (to different data structures)

What would need to be done:
1. `SearchEngine.search()` returns Tracks/Markers (all data is loaded although not everything is used)
    Suggestion: Only return trackId or markerId
2. SearchListActivity gets data directly from `SearchEngine.search()` (everything is in memory)
   Suggestions: user a CursorLoader and ResourceAdapter (like in TrackListActivity)
   Try to use only one query to the database (UNION of Track and Marker) and only load required data.
3. (if possible) do `SearchEngine.search()` computation using SQL on database level

https://developer.android.com/reference/android/database/sqlite/SQLiteQueryBuilder.html#buildUnionQuery(java.lang.String%5B%5D

dennisguse, 2020-11-01:
Note: performance of the current implementation must be really bad.
So, for low-end devices (with a lot of tracks/markers stored), we can really improve here.

dennisguse, 2020-11-01:
We could even go one step further: remove SearchEngine completely as it only reorders the results according to an assumed importance.
From my perspective: a lot of code with very limited actual functionality.

rgmf, 2020-11-02:
> We could even go one step further: remove SearchEngine completely as it only reorders the results according to an assumed importance.
> From my perspective: a lot of code with very limited actual functionality.

I've used it for a couple of times or so. Don't you use this feature? It seems useful, doesn't it?

PS/ I've never read the code and I don't know that part.

dennisguse, 2020-11-02:
I meant to remove the reordering the results and do it on DB level only. The search would still provide the same  results, but the code would be faster and easier.

dennisguse, 2020-11-07:
@rgmf @pstorch I guess, I know the answer, but just to be sure :)
So far our search functionality searched for _tracks and markers_ (in name, description, and category).
What about separating the search functionality?
If search is triggered from the track list, then search for tracks.
If search is triggered from the marker list, then search for markers.

This would simplify our code a lot (no changes to CustomContentProvider), but we lose some functionality.
For marker+track search, I need to create a really relatively complicated SQL union in CustomContentProvider.
And SQLiteQueryBuilder is not helpful (or i did not understood it - very likely ><).

So, what about separating the search functionality?

pstorch, 2020-11-07:
To be honest, I've never used the search function. What can be searched track name? I would need to look into it to give an answer.

rgmf, 2020-11-07:
It looks good for me. I think it makes more sense because I (like user) wouldn't expect markers results when I search in the track list screen.

dennisguse, 2020-11-07:
Thanks for the feedback! Then I will implement the separate search for each of them. That is easy ;)

dennisguse, 2020-11-07:
@pstorch: by name, description, and category (for both).

commit 2faaade7 (2020-11-07): Search is now handled by TrackListActivity and MarkerListActivity.

* data is loaded via cursor,
* removed custom ranking (handled in database),
* search text is wrapped with wildcards, and
* search view can be closed by back button.

Fixes #492.
commit d45ded4a (2020-11-07): Search is now handled by TrackListActivity and MarkerListActivity.

* data is loaded via cursor,
* removed custom ranking (handled in database),
* search text is wrapped with wildcards, and
* search view can be closed by back button.

Fixes #492.
commit fd8b0a51 (2020-11-07): Search is now handled by TrackListActivity and MarkerListActivity.

* data is loaded via cursor,
* removed custom ranking (handled in database),
* search text is wrapped with wildcards, and
* search view can be closed by back button.

Fixes #492.
commit 9ef312fc (2020-11-09): Search is now handled by TrackListActivity and MarkerListActivity.

* data is loaded via cursor,
* removed custom ranking (handled in database),
* search text is wrapped with wildcards, and
* search view can be closed by back button.

Fixes #492.
closed 2020-11-09 by dennisguse