OpenTracksApp/OpenTracks #498: Support for power meters (or smart trainers)

ChristophWurst, 2020-11-06:
**Is your feature request related to a problem? Please describe.**

During off season I cycle mostly indoors on my semi smart turbo trainer *Tacx Satori Smart*. It sends power data via BLE just like a bike power meter (crank/pedals) does.

**Describe the solution you'd like**
I'd like to see a setting for power meters, display of current power on the record page as well as power curves in the reports.

On https://www.bluetooth.com/specifications/gatt/ the sensor is listed as *Cycling Power Profile* and *Cycling Power Service*, so I'm assuming this is also standardized like the cadence and speed sensors.

**Describe alternatives you've considered**
n/a

**Additional context**
IMO a follow-up to the speed/cadence sensor support https://github.com/OpenTracksApp/OpenTracks/issues/16.

dennisguse, 2020-11-06:
I just finished an appointment to test the implementation of https://github.com/OpenTracksApp/OpenTracks/pull/444 with real hardware and I did some small mistakes, but now it works.
At the moment, we only extract the currently applied power nothing else (like cadence, speed etc.) from cycling power sensors.

PS: UI is still a TODO as the power meter UI just lays on to of the cadence UI (which is not hidden if no sensor is connected).

@ChristophWurst 
Just in time :partying_face: 
Can you test this with the mentioned device?

ChristophWurst, 2020-11-06:
Sure, happy to give it a test and report back :)

dennisguse, 2020-11-07:
@ChristophWurst Just one  important note you mentioned a home trainer (which is probably used indoors): while the sensor data will be shown, we only _record_ (store it in the database) when a GPS coordinate is stored.
So, no GPS signal, no diagrams, statistics, export etc.

dennisguse, 2020-11-12:
Powermeter support is implemented, indoor use will be taken care of in #500.

closed 2020-11-12 by dennisguse