OpenTracksApp/OpenTracks #502: Average/max power in track stats

ChristophWurst, 2020-11-07:
**Is your feature request related to a problem? Please describe.**

With #444 the app will be able to read, show and store power values during recording. The track stats overview doesn't reflect the info.

**Describe the solution you'd like**

If a recording contains power data, I would like to see the max and avg power. It could be shown just like the max and avg speed.

**Describe alternatives you've considered**
n/a

**Additional context**

https://github.com/OpenTracksApp/OpenTracks/pull/444#issuecomment-723414756

rgmf, 2020-11-26:
@dennisguse, shouldn't we add new fields in tracks database table in order to get avg/max power, avg/max candence, avg/max heart rate? I ask you this because we have these fields: avgspeed, maxspeed, minelevation, maxelevation and so on.

On the other hand, we can get all that information from trackpoints.

I was going to take this issue but I don't know what should I do, the best option :thinking:

dennisguse, 2020-11-26:
Well....
My guess is that the use case is different - I would argue that the time in different ' power zones'  (like heartrate zones) has a higher relevance.
And then comes the question that those should be configurable (like intervals but in the value domain, e.g.,160-200W: 15min 6s).
So - I would argue against storing them while computing them when needed (like charts/intervals).

@ChristophWurst What do you think?

#185 is related.

rgmf, 2020-11-26:
Okay, I understand... that is more complicated that I thought.

Anyway, apart "power zones" and "heart rate zones" we can show max and avg heart rate too. For power I don't know if that is useful.

hagenw, 2020-11-26:
Regarding power, I would vote for average power and maybe power zones. Maximum power seems to me only meaningful if you are a professional sprinter otherwise it will be dominated by values from starting from 0 km/h.
Power zones are not straightforward as you need to add a configuration option for them or a mechanism that can extract them automatically from your performance.

ChristophWurst, 2020-11-27:
I agree that max power doesn't tell much and average being more meaningful for most athletes. And power zones are indeed not very standardized. But if you let the user configure their FTP you could calculate x zones for them and even approximate the intensity of a session to some degree (avg power / FTP).

However, I guess OpenTracks is not meant to go too much into a training analysis tool. It should rather just collect as much data as possible and export to a format that *Golden Cheetah* or comparable software can read.

closed 2021-02-14 by rgmf