OpenTracksApp/OpenTracks #504: Average power in aggregated stats

ChristophWurst, 2020-11-07:
**Is your feature request related to a problem? Please describe.**

This one is quite similar to #502.

With #444 the app will be able to read, show and store power values during recording. The aggregated track stats overview doesn't use the info. For those people that train with power meters often (e.g. semi-pro cyclists) it might be interesting to see the avg power of all cycling activity. Same applies to running power meters or any other sport where power is relevant.

**Describe the solution you'd like**

If my recording contain items with power data, I would like to see the avg power in the aggregated stats of this sport. It could be shown just like the avg speed.

Maybe there should be some heuristics of how reliable this data source is. Like if only 5% of the recording captured power, the stats would be quite distorted. You don't have this with speed as this metric is always recorded/available.

**Describe alternatives you've considered**
n/a

**Additional context**

https://github.com/OpenTracksApp/OpenTracks/pull/444#issuecomment-723414756

dennisguse, 2021-03-11:
@rgmf We could just add this right?
Or am I missing something?

rgmf, 2021-03-11:
@dennisguse I don't know if that would be feasible.

The average for power must be computed from trackpoints. For one, two or three tracks it'd be feasible but for hundreds of tracks it would get a lot of time, right?

dennisguse, 2021-03-11:
@rgmf You are right, it might be kind of slow (and consume quite some energy).
I don't what the actual use case is and if this could rather be done in another tool like Golden Cheetah.
If it is important, we could start the cache the data after first computation for each track.
But in general: if it makes the code more complicated, I would rather not do it (we have enough trouble with other things ;) ).

@ChristophWurst Do you have some more insight?
Is this still a pain point for you?

ChristophWurst, 2021-03-12:
I do use Golden Cheetah for my long term analysis. When looking at power numbers in OpenTracks or any other tracking software/device my focus is on the power of *one ride*.

Aggregated average power doesn't tell much, so I'd say we can close this.

closed 2021-03-12 by ChristophWurst