OpenTracksApp/OpenTracks #505: Use speed sensor for distance calculation if GPS signal drops

ChristophWurst, 2020-11-07:
**Is your feature request related to a problem? Please describe.**

An idea sparked by https://github.com/OpenTracksApp/OpenTracks/issues/500#issuecomment-723419529: the app currently doesn't record any distance when GPS isn't available. That means when the signal is bad (e.g. driving through a tunnel), no distance will be calculated.

**Describe the solution you'd like**

If the GPS accuracy is bad or the signal drops completely but a speed sensor is available, that sensor data could be used to determine the distance.

**Describe alternatives you've considered**
n/a

**Additional context**

Indoor training #500 with speed sensor only

---

This is really quite an edge case. If you think it doesn't make sense I don' mind if you just close the ticket.

ChristophWurst, 2020-11-09:
With the insights from other tickets I realize that wouldn't be possible right now as the app only records data when GPS is available. So the speed sensor data alone would not produce any recordings. With that in mind #500 would be the prerequisite for this feature.

dennisguse, 2020-11-09:
We could actually store this data in a similar manner like the elevation gain into a column and aggregate until the next GPS point /TrackPoint is inserted.
So, at the end of a trip you would need to get a GPS signal :)
Export and import would also be possible, we would then need to restore the data from extra array instead of the GPS coordinates.
No rocket science here (already done for elevation gain).

commit cbd35d16 (2021-02-13): Use distance from cycling speed sensors.
Fixes #505.
commit 91257c71 (2021-02-14): Use distance from cycling speed sensors.
Fixes #505.
commit 619cfdf9 (2021-03-06): Use distance from cycling speed sensors.
Fixes #505.
commit e37553c3 (2021-03-06): Use distance from cycling speed sensors.
Fixes #505.
closed 2021-03-15 by dennisguse