OpenTracksApp/OpenTracks #508: Bluetooth LE sensors: show name and not just the address

ChristophWurst, 2020-11-09:
**Is your feature request related to a problem? Please describe.**

When I select a sensor, I get a nice list of available sensors. As soon as I select a sensor and the modal closes, only the address of the sensor is shown. As a user I don't care about the address and I certainly don't remember it. If I have more than one sensor of a certain type (let's say two HR straps) then I have to open the sensor picker in order to see the name of the connected device.

[image]

**Describe the solution you'd like**

Show the sensor name (as well as the address) on the settings page.

**Describe alternatives you've considered**

n/a

**Additional context**

n/a

dennisguse, 2020-11-09:
That was done intentionally as the name is not required to connect to a sensor (and therefore no need to store it).
And to switch to another sensor it must be in range - if you want to see the sensor's name (and it is close by), you can open the selection menu...

ChristophWurst, 2020-11-09:
> That was done intentionally as the name is not required to connect to a sensor (and therefore no need to store it).

That makes sense on a technical level.

> And to switch to another sensor it must be in range - if you want to see the sensor's name (and it is close by), you can open the selection menu...

Let me try to rephrase: when I'm not sure which sensor is connection, I not only have to open the settings but also the selection window. If the name of the connected sensor was shown, I'd know in one step whether I need to change the connected device or not.

See my screenshot. Can you guess whether the Garmin cadence sensor is connected or the one of my Tacx trainer? That the kind of UX issue I'm trying to explain.

dennisguse, 2020-11-09:
i see ;)

dennisguse, 2023-12-26:
Finally fixing this issue :rocket: 
Side effect of #1796.

The fun part: Android caches the device names for BLE devices.
And this can be retrieved via the BluetoothGatt object.
So, for Bluetooth connections we need the BluetoothGatt object for a device anyhow and thus get the device name for free.*

* Until it is not cached anymore.

dennisguse, 2023-12-26:
Damn. I need to do the same for the settings...

dennisguse, 2023-12-27:
Now it is done :D

commit 1bd75754 (2023-12-27): Bluetooth sensor settings: show sensor name in addition (if cached via Android OS).

Fixes #508.
closed 2023-12-27 by dennisguse