OpenTracksApp/OpenTracks #512: Automatic export with default track name

slackline, 2020-11-10:
Thanks you for the on-going development of this App, its improving with every iteration.

I was particularly pleased to see the option to automatically export a track in a desired format on completion of recording, saves a lot of time doing it manually.

Having now recorded a couple of tracks with this new feature enabled I see the track is exported with its Number rather than the user specified "Default track name" which in my case I have configured to "Date (ISO8601)".

This is in v3.12.0/v3.12.0-d16cbe04a (Version code 3704).

Would it be possible to have the automatic exports use the "Default track name" as configured?

Thanks and keep up the excellent work.

enilkcals

rgmf, 2020-11-20:
@slackline you can read some discussion about this in #159 

I think is related ;)

slackline, 2020-11-20:
@rgmf Thanks for the pointer.

One of the reasons I prefer ISO 8601 and use it for manual exports is that it facilitates subsequent usage of the GPX files.  If I want to find one for an activity I did on a certain date I can get the from the filename without having to look inside each file.  In that sense ISO 8601 is human readable and adds context compared to track number.

rgmf, 2020-11-20:
@slackline you are right, I have about 200 tracks for now and I cannot find nothing through a File Manager in my computer :sweat_smile:

Someone in #159 pointed out that track's name would be useful in the exported file's name too. It's another possibility. Anyway, #159  is in stand by nowday.

slackline, 2020-11-20:
@rgmf Thanks, the thing with #159 seems to have actually been resolved, manually exporting files we have the option to choose the `Settings > Advanced Settings > Default Track Name > [Date (local) | Date (ISO8601) | Number]`.  I have `Date (ISO8601)` selected and when I export manually the GPX has the ISO8601 filename stub which is great.

I actually am planning on doing a bit more than Importing/Exporting the GPX tracks as I'd like to get a [Panel Dashboard](https://panel.holoviz.org/) written that looks at a directory using [pathlib](https://docs.python.org/3/library/pathlib.html) and reads in all files using the filename (ideally ISO8601) as keys to the dictionary with values as the GPX tracks ([gpxpy](https://pypi.org/gpxpy) makes reading GPX tracks simple). Work and life get in the way but when I get it running I'll share it so others can visualise their workouts.

Anyway, back on topic, as #159 works for manual exports I was surprised that the "Default" filename wasn't used by the "Automatic export".

dennisguse, 2020-11-20:
@slackline We are at the moment overwriting files by filename and the Trackid is an unchangeable property - so it was rather convenient to use for exporting all.
As everything is working nowadays, we can also move to something fancier, more complicated, but actually usable :)

dennisguse, 2021-01-24:
As we have nowadays `opentracks:trackId` in place (UUID to identify a track and not import it if already present in the database), we can do something like: 
`7ceb89d8_BestTrackName.kmz` (i.e., `1stUUID_TrackName.SUFFIX`)

The current implementation trackId (i.e., database identifier) is actually harmful, if one tries to sync multiple devices as those may have the same trackIds. Also importing into a new installation and then exporting will generate different trackIds while the UUID properly random.

slackline, 2021-03-27:
Hi,

Thanks for your continued effort on OpenTracks, its just getting better and better.

I've been trying this out and have under `Settings > Recording > Advanced > Default Track Name` tried all three options `Date (local)` / `Date (ISO8601)` / `Number`.  In all instances when recording of a track ends and the track is saved to disk automatically (`Instant post-workout export` is enabled) the resulting track name is always a number rather than the Date.

I do see some variation in the name of the track as displayed within the application itself between the two settings `Date (local)` / `Date (ISO8601)` but this isn't reflected when exporting automatically.

slackline

dennisguse, 2021-03-27:
@slackline There is no specific option for this.
The filename is just generated differently on export.
But the v3.16.0 is not yet on F-Droid...

sebastianha, 2021-04-11:
I just updated to the latest version and did an export and I personally do not like the new file names.

The reason is that the names begin with the ID which is more or less random. That makes it very hard to find the newest/latest recording.

Perhaps is makes sense to make the file name configurable or put the internal Number the first element of the file name. This would help.

malteger, 2021-04-22:
I second the suggestion made by @sebastianha.

Currently exported files are sorted by UUID as it is first in the name, but date, track number or name would be better in my opinion, as the UUID is of little use to me when directly accessing the files. So I would suggest swapping the positions of the track name and the UUID in the exported file name.

dennisguse, 2022-04-30:
#1207 is ready for testing.

commit bc64a6a3 (2021-03-14): Track export name: first UUID, name and extension (<First 32bits UUID>_<track's name>.<extension>). Fixes #512.
commit 4c402dd4 (2021-03-14): Track export name: first UUID, name and extension (<First 32bits UUID>_<track's name>.<extension>). Fixes #512.
commit bd48944a (2021-03-14): Track export name: first UUID, name and extension (<First 32bits UUID>_<track's name>.<extension>). Fixes #512.
commit cdbf5136 (2022-05-01): Filename template preference: move default to xml and show hint.

Fixes #512.
closed 2022-05-03 by dennisguse