OpenTracksApp/OpenTracks #516: Automatic track export silently fails

pstorch, 2020-11-11:
I was wondering why the automatic track export isn't working on the release version.
Later I connected the device to AS to get a logcat:

```
2020-11-11 20:22:38.992 26923-26923/? W/DocumentsContract: Failed to create document
    android.os.ParcelableException: java.io.FileNotFoundException: Failed to upload document with path /OpenTrack/308.kmz
        at android.os.ParcelableException$1.createFromParcel(ParcelableException.java:82)
        at android.os.ParcelableException$1.createFromParcel(ParcelableException.java:79)
        at android.os.Parcel.readParcelable(Parcel.java:2990)
        at android.os.Parcel.createException(Parcel.java:2083)
        at android.os.Parcel.readException(Parcel.java:2056)
        at android.database.DatabaseUtils.readExceptionFromParcel(DatabaseUtils.java:188)
        at android.database.DatabaseUtils.readExceptionFromParcel(DatabaseUtils.java:140)
        at android.content.ContentProviderProxy.call(ContentProviderNative.java:658)
        at android.content.ContentResolver.call(ContentResolver.java:2049)
        at android.provider.DocumentsContract.createDocument(DocumentsContract.java:1333)
        at androidx.documentfile.provider.TreeDocumentFile.createFile(TreeDocumentFile.java:53)
        at androidx.documentfile.provider.TreeDocumentFile.createFile(TreeDocumentFile.java:45)
        at de.dennisguse.opentracks.util.ExportUtils.getExportDocumentFileUri(ExportUtils.java:88)
        at de.dennisguse.opentracks.util.ExportUtils.exportTrack(ExportUtils.java:38)
        at de.dennisguse.opentracks.util.ExportUtils.postWorkoutExport(ExportUtils.java:31)
        at de.dennisguse.opentracks.services.TrackRecordingService.endCurrentTrack(TrackRecordingService.java:434)
        at de.dennisguse.opentracks.services.TrackRecordingServiceBinder.endCurrentTrack(TrackRecordingServiceBinder.java:69)
        at de.dennisguse.opentracks.services.TrackRecordingServiceConnection.stopRecording(TrackRecordingServiceConnection.java:213)
        at de.dennisguse.opentracks.TrackRecordingActivity$4.onClick(TrackRecordingActivity.java:149)
        at de.dennisguse.opentracks.TrackController$ButtonDelay.lambda$run$1$TrackController$ButtonDelay(TrackController.java:161)
        at de.dennisguse.opentracks.-$$Lambda$TrackController$ButtonDelay$yzFjcE6Sa0ALJEX_mRxG_IsAjAI.run(Unknown Source:2)
        at android.os.Handler.handleCallback(Handler.java:883)
        at android.os.Handler.dispatchMessage(Handler.java:100)
        at android.os.Looper.loop(Looper.java:237)
        at android.app.ActivityThread.main(ActivityThread.java:8016)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:493)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1076)
     Caused by: java.io.FileNotFoundException: Failed to upload document with path /OpenTrack/308.kmz
        at java.lang.reflect.Constructor.newInstance0(Native Method)
        at java.lang.reflect.Constructor.newInstance(Constructor.java:343)
        at android.os.ParcelableException.readFromParcel(ParcelableException.java:56)
        at android.os.ParcelableException$1.createFromParcel(ParcelableException.java:82) 
        at android.os.ParcelableException$1.createFromParcel(ParcelableException.java:79) 
        at android.os.Parcel.readParcelable(Parcel.java:2990) 
        at android.os.Parcel.createException(Parcel.java:2083) 
        at android.os.Parcel.readException(Parcel.java:2056) 
        at android.database.DatabaseUtils.readExceptionFromParcel(DatabaseUtils.java:188) 
        at android.database.DatabaseUtils.readExceptionFromParcel(DatabaseUtils.java:140) 
        at android.content.ContentProviderProxy.call(ContentProviderNative.java:658) 
        at android.content.ContentResolver.call(ContentResolver.java:2049) 
        at android.provider.DocumentsContract.createDocument(DocumentsContract.java:1333) 
        at androidx.documentfile.provider.TreeDocumentFile.createFile(TreeDocumentFile.java:53) 
        at androidx.documentfile.provider.TreeDocumentFile.createFile(TreeDocumentFile.java:45) 
        at de.dennisguse.opentracks.util.ExportUtils.getExportDocumentFileUri(ExportUtils.java:88) 
        at de.dennisguse.opentracks.util.ExportUtils.exportTrack(ExportUtils.java:38) 
        at de.dennisguse.opentracks.util.ExportUtils.postWorkoutExport(ExportUtils.java:31) 
        at de.dennisguse.opentracks.services.TrackRecordingService.endCurrentTrack(TrackRecordingService.java:434) 
        at de.dennisguse.opentracks.services.TrackRecordingServiceBinder.endCurrentTrack(TrackRecordingServiceBinder.java:69) 
        at de.dennisguse.opentracks.services.TrackRecordingServiceConnection.stopRecording(TrackRecordingServiceConnection.java:213) 
        at de.dennisguse.opentracks.TrackRecordingActivity$4.onClick(TrackRecordingActivity.java:149) 
        at de.dennisguse.opentracks.TrackController$ButtonDelay.lambda$run$1$TrackController$ButtonDelay(TrackController.java:161) 
        at de.dennisguse.opentracks.-$$Lambda$TrackController$ButtonDelay$yzFjcE6Sa0ALJEX_mRxG_IsAjAI.run(Unknown Source:2) 
        at android.os.Handler.handleCallback(Handler.java:883) 
        at android.os.Handler.dispatchMessage(Handler.java:100) 
        at android.os.Looper.loop(Looper.java:237) 
        at android.app.ActivityThread.main(ActivityThread.java:8016) 
        at java.lang.reflect.Method.invoke(Native Method) 
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:493) 
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1076) 
2020-11-11 20:22:38.993 26923-26923/? E/TrackRecordingServiceConnection: Unable to stop recording.
    java.lang.NullPointerException: Attempt to invoke virtual method 'android.net.Uri androidx.documentfile.provider.DocumentFile.getUri()' on a null object reference
        at de.dennisguse.opentracks.util.ExportUtils.getExportDocumentFileUri(ExportUtils.java:88)
        at de.dennisguse.opentracks.util.ExportUtils.exportTrack(ExportUtils.java:38)
[... 63 lines cut ...]
        at com.android.server.wm.ActivityTaskManagerService.startActivity(ActivityTaskManagerService.java:1557) 
        at android.app.IActivityTaskManager$Stub.onTransact(IActivityTaskManager.java:1690) 
        at android.os.Binder.execTransactInternal(Binder.java:1021) 
        at android.os.Binder.execTransact(Binder.java:994) 
2020-11-11 20:28:58.969 5862-8527/? W/Bundle: Failed to parse Bundle, but defusing quietly
    android.os.BadParcelableException: ClassNotFoundException when unmarshalling: de.dennisguse.opentracks.content.data.Track$Id
        at android.os.Parcel.readParcelableCreator(Parcel.java:3059)
        at android.os.Parcel.readParcelable(Parcel.java:2981)
        at android.os.Parcel.readValue(Parcel.java:2883)
        at android.os.Parcel.readArrayMapInternal(Parcel.java:3261)
        at android.os.BaseBundle.initializeFromParcelLocked(BaseBundle.java:292)
        at android.os.BaseBundle.unparcel(BaseBundle.java:236)
        at android.os.BaseBundle.getString(BaseBundle.java:1160)
        at android.content.Intent.getStringExtra(Intent.java:8548)
        at com.android.server.wm.ActivityStarter.startActivity(ActivityStarter.java:790)
        at com.android.server.wm.ActivityStarter.startActivity(ActivityStarter.java:694)
        at com.android.server.wm.ActivityStarter.startActivityMayWait(ActivityStarter.java:1925)
        at com.android.server.wm.ActivityStarter.execute(ActivityStarter.java:625)
        at com.android.server.wm.ActivityTaskManagerService.startActivityAsUser(ActivityTaskManagerService.java:1707)
        at com.android.server.wm.ActivityTaskManagerService.startActivityAsUser(ActivityTaskManagerService.java:1603)
        at com.android.server.wm.ActivityTaskManagerService.startActivity(ActivityTaskManagerService.java:1557)
        at android.app.IActivityTaskManager$Stub.onTransact(IActivityTaskManager.java:1690)
        at android.os.Binder.execTransactInternal(Binder.java:1021)
        at android.os.Binder.execTransact(Binder.java:994)
```

pstorch, 2020-11-11:
A manual "Export all" in the same directory works.

dennisguse, 2020-11-11:
1. Could it be that the folder is gone, we wanted to write to?
Is it a local folder or a remote one?

2. Any clue who is trying to unmarshal the Track.Id?
Is this outside of OpenTracks?
Are we sending the Track.Id to the dashboard (which we should not do)?

pstorch, 2020-11-11:
1. The folder exists, the manual "Export all" works with the same directory. It's a NextCloud directory.

2. No clue at the moment. I need to do some more tests. (but not this evening anymore ;)

pstorch, 2020-11-15:
I did some more debugging. Automatic export after recording fails for NextCloud directory, but not for a local device directory. The NextCloud directory works when manually doing a "Export all".

The exception:

```
2020-11-15 21:24:59.472 3285-3285/de.dennisguse.opentracks.debug W/DocumentsContract: Failed to create document
    android.os.ParcelableException: java.io.FileNotFoundException: Failed to upload document with path /OpenTracksDebug/178.kmz
        at android.os.ParcelableException$1.createFromParcel(ParcelableException.java:82)
        at android.os.ParcelableException$1.createFromParcel(ParcelableException.java:79)
        at android.os.Parcel.readParcelable(Parcel.java:2990)
        at android.os.Parcel.createException(Parcel.java:2083)
        at android.os.Parcel.readException(Parcel.java:2056)
        at android.database.DatabaseUtils.readExceptionFromParcel(DatabaseUtils.java:188)
        at android.database.DatabaseUtils.readExceptionFromParcel(DatabaseUtils.java:140)
        at android.content.ContentProviderProxy.call(ContentProviderNative.java:658)
        at android.content.ContentResolver.call(ContentResolver.java:2049)
        at android.provider.DocumentsContract.createDocument(DocumentsContract.java:1333)
        at androidx.documentfile.provider.TreeDocumentFile.createFile(TreeDocumentFile.java:53)
        at androidx.documentfile.provider.TreeDocumentFile.createFile(TreeDocumentFile.java:45)
        at de.dennisguse.opentracks.util.ExportUtils.getExportDocumentFileUri(ExportUtils.java:87)
        at de.dennisguse.opentracks.util.ExportUtils.exportTrack(ExportUtils.java:38)
        at de.dennisguse.opentracks.util.ExportUtils.postWorkoutExport(ExportUtils.java:31)
        at de.dennisguse.opentracks.services.TrackRecordingService.endCurrentTrack(TrackRecordingService.java:434)
        at de.dennisguse.opentracks.services.TrackRecordingServiceBinder.endCurrentTrack(TrackRecordingServiceBinder.java:69)
        at de.dennisguse.opentracks.services.TrackRecordingServiceConnection.stopRecording(TrackRecordingServiceConnection.java:213)
        at de.dennisguse.opentracks.TrackRecordingActivity$4.onClick(TrackRecordingActivity.java:149)
        at de.dennisguse.opentracks.TrackController$ButtonDelay.lambda$run$1$TrackController$ButtonDelay(TrackController.java:161)
        at de.dennisguse.opentracks.-$$Lambda$TrackController$ButtonDelay$yzFjcE6Sa0ALJEX_mRxG_IsAjAI.run(Unknown Source:2)
        at android.os.Handler.handleCallback(Handler.java:883)
        at android.os.Handler.dispatchMessage(Handler.java:100)
        at android.os.Looper.loop(Looper.java:237)
        at android.app.ActivityThread.main(ActivityThread.java:8016)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:493)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1076)
     Caused by: java.io.FileNotFoundException: Failed to upload document with path /OpenTracksDebug/178.kmz
        at java.lang.reflect.Constructor.newInstance0(Native Method)
        at java.lang.reflect.Constructor.newInstance(Constructor.java:343)
        at android.os.ParcelableException.readFromParcel(ParcelableException.java:56)
        at android.os.ParcelableException$1.createFromParcel(ParcelableException.java:82) 
        at android.os.ParcelableException$1.createFromParcel(ParcelableException.java:79) 
        at android.os.Parcel.readParcelable(Parcel.java:2990) 
        at android.os.Parcel.createException(Parcel.java:2083) 
        at android.os.Parcel.readException(Parcel.java:2056) 
        at android.database.DatabaseUtils.readExceptionFromParcel(DatabaseUtils.java:188) 
        at android.database.DatabaseUtils.readExceptionFromParcel(DatabaseUtils.java:140) 
        at android.content.ContentProviderProxy.call(ContentProviderNative.java:658) 
        at android.content.ContentResolver.call(ContentResolver.java:2049) 
        at android.provider.DocumentsContract.createDocument(DocumentsContract.java:1333) 
        at androidx.documentfile.provider.TreeDocumentFile.createFile(TreeDocumentFile.java:53) 
        at androidx.documentfile.provider.TreeDocumentFile.createFile(TreeDocumentFile.java:45) 
        at de.dennisguse.opentracks.util.ExportUtils.getExportDocumentFileUri(ExportUtils.java:87) 
        at de.dennisguse.opentracks.util.ExportUtils.exportTrack(ExportUtils.java:38) 
        at de.dennisguse.opentracks.util.ExportUtils.postWorkoutExport(ExportUtils.java:31) 
        at de.dennisguse.opentracks.services.TrackRecordingService.endCurrentTrack(TrackRecordingService.java:434) 
        at de.dennisguse.opentracks.services.TrackRecordingServiceBinder.endCurrentTrack(TrackRecordingServiceBinder.java:69) 
        at de.dennisguse.opentracks.services.TrackRecordingServiceConnection.stopRecording(TrackRecordingServiceConnection.java:213) 
        at de.dennisguse.opentracks.TrackRecordingActivity$4.onClick(TrackRecordingActivity.java:149) 
        at de.dennisguse.opentracks.TrackController$ButtonDelay.lambda$run$1$TrackController$ButtonDelay(TrackController.java:161) 
        at de.dennisguse.opentracks.-$$Lambda$TrackController$ButtonDelay$yzFjcE6Sa0ALJEX_mRxG_IsAjAI.run(Unknown Source:2) 
        at android.os.Handler.handleCallback(Handler.java:883) 
        at android.os.Handler.dispatchMessage(Handler.java:100) 
        at android.os.Looper.loop(Looper.java:237) 
[... cut ...]
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1076) 

```

Don't know what happens here. 

The BadParcelableException while unmarshalling the Track.Id seems to be caused by the System Classloader, we should use the application Classloader. But I tried several things, like this:
- https://stackoverflow.com/questions/18126249/badparcelableexception-classnotfoundexception-when-unmarshalling-myclass-wh

But it didn't help. I even don't find the place where it happes. In the Stacktrace is no clue about the source or target activity.

pstorch, 2020-11-15:
1 Step further.

I found out that the context which is used in `ExportUtils.exportTrack()` is `ExportService` when using the manual "Export all" function and `TrackRecordingService` when the automatic export is used.
I made a small Proof of Concept to use the `ExportService.enqueue()` method from the `ExportUtils.postWorkoutExport`. I had to provide a ResultReceiver from `TrackRecordingService` for this to work. And it worked. I successfully exported postWorkout a track to the NextCloud directory.

I don't know if the different Context is really the cause, but it works this way. To have the `TrackRecordingService` a ResultReceiver for the `ExportService` might be useful, too. Don't know how to present a error message here at the moment, but that would be a next step.

pstorch, 2020-11-17:
I'll try to find a way to show the export error from the TrackRecordingService.

dennisguse, 2020-11-17:
Can we also use the JobIntentService?

pstorch, 2020-11-17:
> Can we also use the JobIntentService?

That's my plan :wink:

dennisguse, 2020-11-17:
@pstorch We should get into the fortuneteller business :partying_face:

pstorch, 2020-11-18:
Regarding the `BadParcelableException` I think it is from another App which reads the Notifications. When I filter the logcat on OpenTracks only, I don't see it.

commit b2df79ca (2020-11-20): fix #516 post workout export fails silently (#520)

Change the post workout export to use the ExportService (JobIntentService) and directing the user to the SettingsActivity in case the export failed to check the export directory.
closed 2020-11-20 by pstorch