OpenTracksApp/OpenTracks #523: Crash on start recording in v3.13.0

dennisguse, 2020-11-21:
**Describe the bug**
Crash on record start

**To Reproduce**
1. Press record

If applicable:
```
2020-11-21 09:46:13.080 19606-19606/de.dennisguse.opentracks E/AndroidRuntime: FATAL EXCEPTION: main
    Process: de.dennisguse.opentracks, PID: 19606
    android.content.res.Resources$NotFoundException: Drawable de.dennisguse.opentracks:drawable/ic_track_recording with resource ID #0x7f0800ab
    Caused by: android.content.res.Resources$NotFoundException: File res/drawable/ic_track_recording.xml from drawable resource ID #0x7f0800ab
```

**Technical information**
 - Version: v3.13.0 F-Droid and PlayStore

dennisguse, 2020-11-21:
The resource seems to be removed by 
```
            shrinkResources true
            minifyEnabled true
```
as it is only referenced from Java not the XML.

Quick fix: remove shrink/minify.
Real fix: define the button state in a drawable.

dennisguse, 2020-11-21:
PlayStore has new version.
For F-Droid we have to wait.

slackline, 2020-11-25:
Just got the update to v3.13.2 via F-Droid this morning, thank you.

dennisguse, 2020-11-25:
Sorry for the inconvenience. 
Note to myself: be really careful while releasing UI changes....

closed 2020-11-25 by dennisguse