OpenTracksApp/OpenTracks #527: Running Speed varies from Intervals to By Time

jdohm, 2020-11-22:
**Describe the bug**
While watching the Data from a previous run, I noticed that the speed shown on the "Intervals" tab doesn't match the graphics shown on "By Time" and "By Distance".

**To Reproduce**
1. Record a Run
2. Click on the saved track
3. switch to intervals tab
4. switch to by time tab and compare

**Screenshots**
While the speed by interval is displayed as ~10 km/h the graphic shows speeds around 9km/h. 
[image]
[image]

**Technical information**
 - Device: LG G6 (H870)
 - OS: LineageOS 17.1
 - Version v3.12.0

Thanks for your nice Work.
If I simply missed something please let me know.

dennisguse, 2020-11-22:
Yes you are right - the computation is done in a different manner.

The interval is based upon distance / time: https://github.com/OpenTracksApp/OpenTracks/blob/679c1d7e3e326ec0400b850e291373d6c7612301/src/main/java/de/dennisguse/opentracks/viewmodels/IntervalStatistics.java#L117

The diagrams use the `TrackStatisticsUpdater` to recompute the `TrackStatistics` (the data shown on the first tab) and here a smoothing is applied: https://github.com/OpenTracksApp/OpenTracks/blob/0d493c410e31fffb1e39be210d254c4aa3c2a98f/src/main/java/de/dennisguse/opentracks/stats/TrackStatisticsUpdater.java#L51

So, the diagram is more like the data was that shown while recording and the interval tab is more "raw".

A related issue is: https://github.com/OpenTracksApp/OpenTracks/issues/213

jdohm, 2020-11-23:
Thanks @dennisguse 

Maybe a solution would be to increase the GPS accuracy and turn of the smoothing. 
In my case, I manually calculated the average speed to be 9.75. 
This would be below the "Intervall" but above the "By Time" value. 

I'm going to test a higher GPS accuracy setting tomorrow and will report back. 
I know it's more complicated than this, not everyone has the same phone. 
Maybe this information is valuable anyway.

dennisguse, 2020-11-23:
We handle GPS accuracy by just ignoring every GPS fix that is above the set limit. So, we store less data and this can be beneficial or not.

dennisguse, 2023-06-07:
This shouldnt be a problem anymore as speed is not filtered.

closed 2023-06-07 by dennisguse