OpenTracksApp/OpenTracks #535: OpenTracks crashes immediately after starting recording

Saboe, 2020-11-28:
When I start OpenTracks and want to record a track, OpenTracks crashes immediately. It seems to be a new bug with version 3.13.0. With prior versions there wasn't this issue.

I get only the following crash report/screen from Adnroid:
OpenTracks keeps stopping
(i) App info
X Close app

**Technical information**
 - Device: Fairphone 3
 - OS: Android
 - Version 10, Build 8901.3.A.0054.20200929

dennisguse, 2020-11-28:
Sorry for the inconvenience.
You can just upgrade to v3.13.2 that contains fixes for this.
Fixed in #513.

closed 2020-11-28 by dennisguse

Saboe, 2020-11-28:
Thanks for the hint. Yesterday I've checked F-Droid for the latest version and it shows version 3.13.0 but now it's 3.13.2. Now I've upgraded OT.