OpenTracksApp/OpenTracks #537: Ignore deprecated statistics waypoint on import

pstorch, 2020-11-28:
It still bugs me that I cannot import my old tracks.
I once tried to ignore invalid coordinates, but then the discussion on how to handle the pause/resume trackpoints in the database and the im-/export files came up. See #316 

Pause and Resume use latitude 100 and 200 with longitude 0, but the invalid coordinates from my old kmz exports contain latitude 100 and longitude 180. So I dig a little bit deeper. From the kmz file I saw it is a Placemark with "Split 0" in the name:

```
<Placemark>
<name><![CDATA[Split 0]]></name>
<description><![CDATA[Total distance: 0.00 km (0.0 mi)
Total time: 00:00
Moving time: 00:00
Average speed: 0.00 km/h (0.0 mi/h)
Average moving speed: 0.00 km/h (0.0 mi/h)
Max speed: 0.00 km/h (0.0 mi/h)
Average pace: 0:00 min/km (0:00 min/mi)
Average moving pace: 0:00 min/km (0:00 min/mi)
Fastest pace: 0:00 min/km (0:00 min/mi)
Max elevation: -9223372036854775808 m (-9223372036854775808 ft)
Min elevation: 9223372036854775807 m (9223372036854775807 ft)
Elevation gain: 0 m (0 ft)
Max grade: 0 %
Min grade: 0 %
Recorded: 6/23 11:29
]]></description>
<TimeStamp><when>1970-01-01T00:00:00Z</when></TimeStamp>
<styleUrl>#waypoint</styleUrl>
<Point>
<coordinates>180.0,100.0</coordinates>
</Point>
</Placemark>
```

Newer files don't contain this Placemark anymore.

Then I digged a little bit in the git history and found this commit:
https://github.com/OpenTracksApp/OpenTracks/commit/09d960c7a1228e4822410abedf3cef9e45440ef5

So this is not used anymore. 

Can we ignore those Markers while importing? Markers are not relevant for the Track itself, right? So ignoring markers with invalid coordinates wouldn't hurt. If someone thinks we should still stop the import on invalid marker coordinates, I can offer to only ignore those obsolete statistic markers.

dennisguse, 2020-11-28:
Now I remember: these are old markers to store statistics.
I removed those ages ago, so yes - let's ignore them.

pstorch, 2020-11-28:
Can we ignore all markers with invalid coordinates or only those statistic markers?

commit f5161093 (2020-11-29): fix #537 ignore waypoint markers with invalid

coordinates on import.

I've also added a unit test with such an export file.
commit 215c28be (2020-11-29): fix #537 ignore waypoint markers with invalid

coordinates on import.

I've also added a unit test with such an export file.
closed 2020-11-29 by pstorch