OpenTracksApp/OpenTracks #540: Android app crashing

matthead86, 2020-12-01:
**Describe the bug**
Since the most recent update the app closes when the 'GPS started' notification popup appears. If I click on the notification, then I get to the main recording page. If I miss the notification, I can't access that page. Reopening the app and restarting the record crashes the app. If I try the app switcher I still can't access the live recording screen. 

**Technical information**
 - Device: Nokia 7 Plus
 - OS: Android 10
 - Version V3.13

dennisguse, 2020-12-01:
Sorry.
Can you please try to upgrade to v3.13.2?
I hope that you only experience the same issue described here: https://github.com/OpenTracksApp/OpenTracks/issues/523

matthead86, 2020-12-01:
Yes. 3.13.2 fixes the issue. Thanks.

Keep up the good work!

closed 2020-12-01 by dennisguse