OpenTracksApp/OpenTracks #545: Use ThumbnailUtils for marker's photos

rgmf, 2020-12-06:
**Is your feature request related to a problem? Please describe.**
Marker's list has trouble when there are several large photos.

**Describe the solution you'd like**
Use ThumbnailUtils (see https://github.com/OpenTracksApp/OpenTracks/pull/519#issuecomment-729231165)
https://developer.android.com/reference/android/media/ThumbnailUtils

rgmf, 2020-12-07:
Seem that using ThumbnailUtils.extractThumbnail can be problematic because you are actually loading in memory a copy of the large bitmap, so this is not a good way of managing big images.

In this official doc explain how to do it:
https://developer.android.com/topic/performance/graphics/load-bitmap?hl=es

rgmf, 2020-12-08:
Seem that the problem comes from the moment where image is loaded (from storage). The size of the image is not the problem (altought it doesn't help).

Cache strategy could be the solution: https://developer.android.com/topic/performance/graphics/cache-bitmap

dennisguse, 2020-12-10:
We could also create thumbnails and store those beside the actual pictures.

rgmf, 2020-12-11:
> We could also create thumbnails and store those beside the actual pictures.

Yes, I thought on that possibility but the actual issue is that we have to load photo's file on a separate thread. Doesn't matter if photo is large or small in size: the marker's list is really slow if images are not loaded in a separate thread.

So, I think the best approach if what I've proposed in #546

commit 48162628 (2020-12-18): Creates thumbnail from marker's photos for marker's list.
These thumbnails are created and loaded async.
Fixes #545.
commit f20b74d3 (2020-12-19): Creates thumbnail from marker's photos for marker's list.
These thumbnails are created and loaded async.
Fixes #545.
commit 9b6acb0d (2021-01-24): Creates thumbnail from marker's photos for marker's list.
These thumbnails are created and loaded async using cache from free memory available.
Fixes #545.
commit 8a08e7b2 (2021-01-24): Creates thumbnail from marker's photos for marker's list.
These thumbnails are created and loaded async using cache from free memory available.
Fixes #545.
closed 2021-01-24 by rgmf