OpenTracksApp/OpenTracks #555: Elevation Gain is 0?

pdhixenbaugh, 2020-12-26:
**Describe the bug**
Circuit track has elevation graph, but shows zero elevation Gain overall and in intervals

**To Reproduce**
Import GPX

If applicable:
* add screenshots
[image]
[image]
[image]

**Technical information**
 - Device: Pixel 3
 - OS: Android 11
 - Version 3761

dennisguse, 2020-12-26:
Can you export a recorded track (w/ gain) as KML and import it again?
It might be that we did not implement the export of the gain (got via barometric sensor) for GPX.

dennisguse, 2020-12-26:
Just checked the code and, yes, we do not export/import the gain data with GPX.
https://github.com/OpenTracksApp/OpenTracks/blob/main/src/main/java/de/dennisguse/opentracks/io/file/exporter/GpxTrackWriter.java

rgmf, 2020-12-26:
> Just checked the code and, yes, we do not export/import the gain data with GPX.
> https://github.com/OpenTracksApp/OpenTracks/blob/main/src/main/java/de/dennisguse/opentracks/io/file/exporter/GpxTrackWriter.java

@dennisguse I noticed that when I was working on "elevation loss" but I thought that there was a reason. I cannot remember why we didn't add this feature for GPX files.

@pdhixenbaugh Was GPX file you are importing exported using OpenTracks?

@dennisguse Could we build "elevation_gain" and "elevation_loss" for "trackpoints" from elevation in GPX/KML files when these files come from another software?

dennisguse, 2020-12-26:
No real reason (I guess); we always exported less to GPX and that was it.
About importing from another software: I guess not, but I also don't expect KML to work. We are exporting data into a custom extension. I am not really sure if we even could import speed (exported in a similar way). Usually, gain/speed are derived from the GPS data alone (the actual coordinates).

commit 2b7fb2dd (2020-12-26): Added opentracks:gain and opentracks:loss information in GPX files. Fixes #555.
commit 284abd7d (2020-12-26): Added opentracks:gain and opentracks:loss information in GPX files. Fixes #555.
commit fd1b7b8a (2020-12-26): Added opentracks:gain and opentracks:loss information in GPX files. Fixes #555.
closed 2020-12-26 by rgmf