OpenTracksApp/OpenTracks #558: offer "Open" in GPX import activity

vanous, 2021-01-03:
- [x] I searched the issue tracker and didn't find anything related.

Thank you for all the work and Happy new year! :)

When importing GPX from an intent, it would be very nice to be able to (conditionally) open OpenTracks right away. One way to do this could be if the OK button only takes part of the screen and the other part is taken by "Open". 

| OK | Open |

As "Open" is ambiguous, question could be whether "open app" or "open track" is meant. In fact, the user might be interested in any of the three (not sure if identification of the currently imported track is available to be passed).

So my proposal would be to split the screen horizontally into 3 buttons, which would offer:

| Done | Open app | Open track |

cheers
P.

dennisguse, 2021-01-03:
I don't get it - can you attach a screenshot of the current implementation?
And I am also not sure that I understood the proposed solution as I don't know the problem this would solve for you ;)

PS: Happy new year too ><

vanous, 2021-01-03:
Oh, sure, sorry.

First i select a file for export and choose OpenTracks:

[image]

Then the GPX is imported, here, instead of the OK, i would propose the 

Done | Open app | Open track

[image]

dennisguse, 2021-01-03:
I see ;)
In case, it is only one track: we can add a button to open the track, right?
Anything else?

vanous, 2021-01-03:
Well, that would be awesome (low priority also). As i mentioned, i think it would be nice to keep the previous option just to dismiss (Done). My case was to import one track and look at several tracks, but i agree that having just "Done" and "View" ("Open") is sufficient to get to the app and continue from there :)

commit 7b67d607 (2021-01-10): ImportActivity offers to open track option (when only one track is imported). Fixes #558.
commit 9b974434 (2021-01-10): ImportActivity offers to open track option (when only one track is imported). Fixes #558.
commit 9fbc3c96 (2021-01-10): ImportActivity offers to open track option (when only one track is imported). Fixes #558.
commit 860df019 (2021-01-10): ImportActivity offers to open track option (when only one track is imported). Fixes #558.
closed 2021-01-10 by rgmf