OpenTracksApp/OpenTracks #559: import multiple GPX files via intent at once

vanous, 2021-01-04:
Importing multiple GPX files at once would eventually be a nice addition. So far i have only researched and tested the manifest changes, as per snippet below. `ImportActivity` would need to receive multiple files most probably via `getParcelableArrayListExtra`, but i didn't manage to research this further.

```xml
<intent-filter>
    <action android:name="android.intent.action.SEND_MULTIPLE" />
    <category android:name="android.intent.category.DEFAULT" />
    <data android:mimeType="application/gpx+xml" />
</intent-filter>
<intent-filter>
  <action android:name="android.intent.action.SEND_MULTIPLE" />
  <category android:name="android.intent.category.DEFAULT" />
  <data android:mimeType="application/gpx" />
</intent-filter>
```

dennisguse, 2021-03-11:
@vanous do you still need this?

vanous, 2021-03-13:
Hi @dennisguse :)

personally, due to the lockdown and limited allowed movement range (affecting my possible explorations), i had been focusing my activity related development in Gb on a GPS-less activity detection, but once hiking in hills is allowed again, GPS usage will go up for me. From some feedback i see in our chatroom, people do use the GPS, so my request was coming from me as a user but also from the larger vision of allowing more free data flow between apps.

Feel free to close with "looking for help" :)

PR #649 (2021-03-14): Import: allow import multiple select files.
Fixes #559.

@vanous Could you check that this works as you expect it to work?
commit 503cba13 (2021-03-14): Import: allow import multiple select files.

Fixes #559.
closed 2021-03-14 by dennisguse