OpenTracksApp/OpenTracks #562: New screenshots for F-Droid/PlayStore

dennisguse, 2021-01-08:
We had quite some UI changes (interval etc), so updating screenshots might be a good idea.

@pstorch I was thinking about including two screenshots with OSMDashboard (one OSMDashboard in fullscreen and one as free floating overlay), so that potential user get aware of it.
And we would need to show it's name somehow.
What do you think?

pstorch, 2021-01-08:
Good idea

rgmf, 2021-01-24:
@pstorch @dennisguse What do you think of doing something like this? With this kind of images we can introduce some text explaining the differents parts of the app (a mini how-to) also:

[image]

tarah20202, 2021-01-24:
Hello sir tank you for the massage sir. I don't know wat is this?? This is
app?? Sir I want clipclaps modded.

closed 2021-01-29 by rgmf