OpenTracksApp/OpenTracks #563: Shortened text in translations

FTno, 2021-01-09:
[image]

As seen in the screenshot above from the Settings some text is shortened (this also happens elsewhere). I think this may be an issue in other languages as well and suggest to reduce the margins, especially the left one.

BTW 'Prevent re-import of tracks' as can be seen in the screenshot, is not translatable.

dennisguse, 2021-01-09:
It might be that we cannot do anything about the shortening as this is a Android default component.
Let me check; only other option is to use shorter texts (in all translations).

"prevent re-import" can now be translated; it was marked as not translatable.

dennisguse, 2021-01-17:
I did some digging and preference titles are by default a single line only leading to `...`.
In fact, it depends on the screen size vs. font size vs. language and usually preference titles are rather short.
And the rest of the list (margins etc) is defined by Android itself.

@FTno any chance that we can use a different (aka shorter) term for nb?

FTno, 2021-01-17:
As the margins cannot be reduced, I will try to shorten terms instead.

closed 2021-01-17 by FTno