OpenTracksApp/OpenTracks #566: Long activity's name and sensor data on StatisticsRecorded

rgmf, 2021-01-16:
**Is your feature request related to a problem? Please describe.**
When user set a long name for his/her activities then s-he cannot read it on any of the views anymore, either ListActivity nor StatisticsRecorded.

Also, OT records sensor data that user cannot see anymore.

**Issue's screenshot**
[image]

**Describe the solution you'd like**
I think we should redesign, at least, the StatisticsRecorded layout. I've been playing with this layout and I show you one option that I don't like anything, but you could understand more or less what I'd do:

[image]

**Describe alternatives you've considered for issue name**
I think after recording, in the recorded statistics, we should show to the user meta-information like the complete name, description (if any), start and end datetime... all this information is useful.

**Describe alternatives you've cosidered for issue sensor**
By the way we should add sensor data on StatisticsRecorded to complete it. At least:
- Heart rate: max and avg (https://github.com/OpenTracksApp/OpenTracks/issues/525).
- Power: avg power (useful regarding some comments in here https://github.com/OpenTracksApp/OpenTracks/issues/502)

All we need to do is query trackpoints database table to get all this information and load them on StatisticsRecorded.

dennisguse, 2021-01-16:
Sounds good!

PS: Regarding to changes TrackStaticstics; I need to merge #553 first.
Too many changes there... I am hopefully done by tomorrow.

rgmf, 2021-01-16:
> PS: Regarding to changes TrackStaticstics; I need to merge #553 first.

Yes, yes, I don't want to touch anything before #553... but no hurry :grinning:

commit a22f845e (2021-03-06): Statistics recorded fragment shows activity's information (name, description, start datetime).
Re-arrange stats putting the focus on moving stats.
Fixes #566.
commit 74fcbc6b (2021-03-09): Statistics recorded fragment shows activity's information (name, description, start datetime).
Re-arrange stats putting the focus on moving stats.
Fixes #566.
commit 34eb7a83 (2021-03-12): Statistics recorded fragment shows activity's information (name, description, start datetime).
Re-arrange stats putting the focus on moving stats.
Fixes #566.
closed 2021-03-12 by rgmf