OpenTracksApp/OpenTracks #571: Retry option on instant workout export error

pstorch, 2021-01-21:
I have the instant workout export enabled. To a Nextcloud folder.
Usually I stop the workout when I reach my home. This is also the point when my device tries to connect to my home wifi.
It happens quite often to me that OT tries to export the track when the wifi connection is not yet properly working, so the instant export fails. Two to three seconds later it is working.
So it would be very handy to have a retry button on that error dialog.

dennisguse, 2021-01-22:
Would a one time notification with a retry option help?

pstorch, 2021-01-22:
I don't think a notification is necessary. We have this new dialog which informs about the failed export. Usually you have OT still open, because you just stopped the track recording. At the moment this dialog has only an ok button. It would be convenient to also have a retry button here.

pstorch, 2021-01-31:
I looked into it, but I doubt it is an easy change. The error is reported to the TrackRecordingService and forwarded to the SettingsActivity which shows this dialog. There it is too late to retry the export, I guess.

rgmf, 2021-01-31:
> I looked into it, but I doubt it is an easy change. The error is reported to the TrackRecordingService and forwarded to the SettingsActivity which shows this dialog. There it is too late to retry the export, I guess.

Why is too late to re-export? SettingsActivity can assume that the error comes from the last track inserted in the database, right?

pstorch, 2021-01-31:
We could also forward the trackId via Intent Extras. I was just not sure if it should be the responsibility of the SettingsActivity to actually handle the retry or enqueue it again.

rgmf, 2021-01-31:
Yes, you are right, seem odd that SettingsActivity take on that responsibility.

dennisguse, 2021-01-31:
Why is the SettingsActivity at all involved? Sounds more like ExportActivity.

pstorch, 2021-01-31:
Hmm, was it my idea? 🤔
I think it was because most probably the export directory is not right or the grant not given anymore, so the user can check in the Settings Activity right away this folder or assign a new one.

dennisguse, 2021-07-29:
@pstorch Is this still an issue?

pstorch, 2021-07-29:
It's happening very seldom, so not an issue for me. If you want to do some housekeeping, feel free to close it.

closed 2021-07-29 by dennisguse