OpenTracksApp/OpenTracks #575: Increase contrast between buttons and background during tracking

dapgo, 2021-01-24:
During outdoor sport I've found problems identifying buttons "record, pause, stop" and their current status.

As background color in the buttons area is black, the red color or grey have little contrast and specially if the screen is using tones of grey (used by the "ultra saving mode" of phones as Samsung) but also for sunglasses.  

See below the aspect and colors in ultra saving mode enabled.
[image]
[image]

dennisguse, 2021-01-24:
Any suggestions how to adapt for this?
PS: In LineageOS, one can activate monochrome mode in the developer options.

rgmf, 2021-01-24:
Doing the circle, square and two lines from record, stop and pause buttons a little bit greater the contrast will become greater too. I don't know if it would be enough to distingush between record, stop and pause but I think so.

paulle, 2021-03-05:
I would supprt this to have the buttons with higher contras (dark theme). Replacing the dark red with brighter one and the grey with green or yellow. As it is it's hard to see when one is outdoor.

paulle, 2021-03-05:
[image]

dennisguse, 2021-03-06:
@paulle You can also use the light theme during the day (it is in the settings).
We need something more holistic than just changing the border of the buttons.

closed 2022-05-08 by rgmf