OpenTracksApp/OpenTracks #584: Customizable notification

mishari, 2021-01-31:
**Is your feature request related to a problem? Please describe.**
When I'm running I would like to be able to see vital statistics at a glance through the phone's notification bar instead of having to unlock my phone and going to the app. In my case, I would like to see total distance and total time instead of GPS accuracy.

**Describe the solution you'd like**
I would like to see settings that allows me to pick what is displayed in notifications.

**Describe alternatives you've considered**
Alternatively just a hard coded display of distance will suffice for my needs.

**Additional context**
I'm happy to attempt to write it if you can provide some hints as to where / how this can be done.

dennisguse, 2021-01-31:
We have an issue for this: https://github.com/OpenTracksApp/OpenTracks/issues/483
Main use case is to show this kind of info on a smartwatch.

If you want to code:
Take a look at the `NotificationManager`.
At the moment, it gets a `TrackPoint` (contains location, speed, and might also sensordata) when a new location is provided from the LocationManager (via `TrackRecordingService`).
The info you want is in the `TrackStatistics` (also available in the `TrackRecordingService`).

PS: In the settings you find an option to show the OpenTracks statistics (while recording) on the lock screen (if the app was focused when locking the device).

closed 2021-03-18 by rgmf

dennisguse, 2021-03-18:
@vanous Out of curiosity (and lack of test devices like Miband): is notification shown on a wearable device and does it show the latest information?

vanous, 2021-03-18:
I can try tomorrow. I block all notifications but I will allow ot and will report.

vanous, 2021-03-19:
I tried this. This is an ongoing notification, right? These are blocked in Gadgetbridge in general.

dennisguse, 2021-03-19:
@vanous Yes, it is. The notification is created by a ForegroundService (to keep it alive) and then updated whenever we get a new GPS fix.
Any chance to enable this in Gadgetbridge as it might be a really nice feature with the new information while doing sports etc.

vanous, 2021-03-20:
> @vanous Yes, it is. The notification is created by a ForegroundService (to keep it alive) and then updated whenever we get a new GPS fix.
> Any chance to enable this in Gadgetbridge as it might be a really nice feature with the new information while doing sports etc.

I will check it out. There are several things and protections in place already that will need to be looked at.

vanous, 2021-03-29:
This has been merged and we tagged yesterday, now waiting for F-Droid to build :)

https://codeberg.org/Freeyourgadget/Gadgetbridge/pulls/2236

It is advisable to set "Minimum time between notifications" to a few seconds, because depending on how often the notifications are updated, battery can suffer and firmware might get overwhelmed :)

I still need to mention this in our wiki...

dennisguse, 2021-03-29:
@vanous Pretty awesome! That is really cool!

@rgmf Might be interesting for you!

rgmf, 2021-03-29:
@vanous thank you so much!!!

@dennisguse of course it is :smile:

commit 96dfb65d (2021-03-13): Notification manager shows GPS accuracy, distance and speed. Fixes #584 and #483.
commit bc660e03 (2021-03-18): Notification manager shows GPS accuracy, distance and speed. Fixes #584 and #483.
commit 6d926ca7 (2021-03-18): Notification manager shows GPS accuracy, distance and speed. Fixes #584 and #483.