OpenTracksApp/OpenTracks #588: Heart rate export problem

neidert, 2021-02-04:
OpenTracks, version v3.14.3, Samsung S 8, Android 9

I run with LE Bluetooth heart rate measurement. When I export the track (for example to Locus Map or Garmin, as gpx), the heart rate data are no longer exported or displayed on Garmin, Locus Map ...
When I edit the OpenTracks-gpx-file, however, I see the heart rate entered:
<gpxtpx:hr>95</gpxtpx:hr>
I believe: This error has existed since the penultimate version.
Thank you

dennisguse, 2021-02-04:
Thanks for the report!

I just checked the code and it looks okay: https://github.com/OpenTracksApp/OpenTracks/blob/f8f78d28dd093165ff0da8537eaf8988cd8f5f88/src/main/java/de/dennisguse/opentracks/io/file/exporter/GpxTrackWriter.java#L207

dennisguse, 2021-02-05:
@neidert We need some more info to figure out if there is a problem.
Please reopen, if this problem persists.

closed 2021-02-05 by dennisguse

neidert, 2021-02-06:
Here is the link to the file of a run with heart rate measurement:
https://neidert-wg.webo.family/index.php/s/xH3sgcSxNMdXcnN

- OpenTracks graphic "by time" by track" shows the heart rate of this track
- GPX viewer online displays heart rate from this track: https://www.bernhard-gaul.de/gpxviewer/gpxviewer.php
- Upload the same gpx file to Garmin connect or Locus Map or Runalyze.com does not display any heart rate data

What more information do you need?

Thank you very much
neidert

rgmf, 2021-02-06:
@dennisguse @neidert if you change:

```xml
xmlns:gpxtpx="http://www.garmin.com/xmlschemes/TrackPointExtension/v2"
```
by

```
xmlns:gpxtpx="http://www.garmin.com/xmlschemes/TrackPointExtension/v1"
```

Only downgrade the GPX TrackPointExtension from version 2 to version 1 the heartrate works, at least on Strava (I've created a fake file with fake heart rate to test it).

@neidert can you change `v2` to `v1` and check it out on Garmin, Locus Map...?

What a mess!!! :confused: 

PS/ Hunted here: https://help.locusmap.eu/topic/hrm-data-not-visible-in-garmin-connect

neidert, 2021-02-06:
@rgmf 
I changed the gpx-File to xmlns:gpxtpx="http://www.garmin.com/xmlschemes/TrackPointExtension/v1"
When I import this changed file to Locus Map (pro), the heartrate doesn´t work.
When I directly import th gpx file to Runalyze, it works with version v2, but not by this way: Export to Locus map (v2 or v1) , from there automatical export to Runalyze.
Thanks!
neidert

dennisguse, 2021-02-06:
@rgmf Great work!

PS: How do we document such stuff?
It is kind of hard to understand those formats/interoperability etc and knowing all these tiny, nasty details.

rgmf, 2021-02-07:
> PS: How do we document such stuff?
> It is kind of hard to understand those formats/interoperability etc and knowing all these tiny, nasty details.

:thinking: maybe we should add documentation for all these stuff... with a simple Markdown file inside doc explaining formats accepted, opentracks extension, drawbacks and all related to import/export file formats.

Were you thinking on something like that?