OpenTracksApp/OpenTracks #590: Autostart for both, the app and the tracking-function

Elmi77, 2021-02-06:
**Is your feature request related to a problem? Please describe.**
I'm using the app in my car with my Android car radio and whenever I start the car, I manually have to open the app, start the tracker and select the graph I want to see (speed and height over time). This is time-consumign and sometimes I forget it.

**Describe the solution you'd like**
Ass the app is a nice software to track the driven routes, I would like to have it started whenever the car is stared/the radio is turned on:
- auto-start the app on quick-boot of Android
- start tracking
- go to the last view/tab automatically

dennisguse, 2021-02-06:
Adding quick setting tile to start a recording is great idea!

But I guess that moving to a specific tab is too much as not everybody wants to see the same.
Then we need some configuration etc.
This adds just too much complexity.

The implementation should be straight forward using a TileService.
Then we can call OpenTracks `TrackRecordingService.startRecording()`.

https://developer.android.com/reference/android/service/quicksettings/TileService

PS: I am stuck with GPX/KML stuff, so if anybody wants to grab that: have fun.

dennisguse, 2021-02-07:
Another option would be to implement app shortcuts. See #331

Elmi77, 2021-02-07:
Moving to a specific tab should not be that complicated. No separate configuration is needed, the app could just remember what tab was opened last and then on next start go back to this tab.

dennisguse, 2022-01-09:
We will build something using some external "if this, then that"-tool #671.
Don't know if this will include the tab selection, but recording  start/stop will be included.
Ongoing PR #1072 .

closed 2022-01-09 by dennisguse