OpenTracksApp/OpenTracks #593: Polar H10 not detected by OpenTracks on a Huawei P40 lite

hg-runs, 2021-02-08:
**Describe the bug**
Tried to use a Polar H10 heart rate monitor (which is listed as usable by OpenTracks).
It's detected by the smartphone but not within the app.

**Am I missing something here?**

**Screenshots**
[image]
[image]
[image]

**Technical information**
 - Device: Huawei P40 lite
 - OS: Android version 10

dennisguse, 2021-02-08:
Nice detailed report!

TLDR: just try it again with the H10 on your body.
The problem is that the H10 is turned off (aka powersaving mode).
OpenTracks does not check which sensors are paired with your phone, but searches for sensors how are actively announcing that they are there.
And the H10 only turns on, when it is attached to the body (i.e., it's sensor have contact with something that is conductive).

Other thing that might happen: the battery of the H10 is empty.

PS: I used an H10 to implement the HR functionality :)

closed 2021-02-08 by dennisguse

hg-runs, 2021-02-10:
Thanks for the quick answer.

Back with some more information:

- The H10 was new. I've changed the battery though but no luck
- The H10 was not paired
- The H10 was attached to my body - other wise it would not have been listed in the bluetooth discovery of the phone

Why the smartphone's bluetooth discovery is working but not the app's one?

Obviously I'm interested to get this running.
Anything else I can test/do?

dennisguse, 2021-02-10:
Can you check that either FitoTrack or RunnerUp can access the HR sensor?

Some more details:
In the settings, OpenTracks searches for (HR) sensors.
This means that the sensors must be turned on, not connected to any device (otherwise they don't announce that they are available), and in their announcement package announce that the HR gatt is provided.
A Miband doesnt do the later and we have a workaround in place for this.

And we tell Android to search for HR sensors (filter by serviceId): https://github.com/OpenTracksApp/OpenTracks/blob/e8017a4e9d8c4c420d9c1663dc40a4a036d77f49/src/main/java/de/dennisguse/opentracks/settings/bluetooth/BluetoothLeSensorPreference.java#L184

hg-runs, 2021-05-05:
Just FYI:
Using this version of OpenTracks the Polar H10 all of sudden is recognized on a Huawei P40 lite.
But not only the HRM but also a GARMIN speed sensor 2 and a cadence sensor 2 (which weren't working with any other previous versions of OpenTracks).

Neither sure what has been changed nor had a look at the repository code. Fitotrack and RunnerUp are still not recognizing the Polar H10
[image]

[image]

dennisguse, 2021-05-05:
@hg-runs Great!
i would really love to know why as I don't recall any change.
Whatever - glad that it is working and really cool that you use all these sensors!