OpenTracksApp/OpenTracks #595: Using pl settings android pronounces extras strange "ygreek" (Upsilon)  between minutes and seconds

matkor, 2021-02-09:
Using pl settings android pronounces extras strange _"ygreek"_ between minutes and seconds

**To Reproduce**
Start activity, wait for any period TTS information in polish.
I think it is using:
<string name="voice_template">Moje trasy: %1$s w czasie %2$s z prędkością %3$s</string>
but have no idea what %2$s is filled with, but is pronounced:
_"foo minut ygreek bar sekund"_
instead of just:
_"foo minut bar sekund"_
or
_"foo minuty bar sekundy"_
(depending on last digit of numeral )

Same for lap times.

**Fix guess**
Perhaps (if that is template used)  "(y)"  should be removed from templates like:

    <strring name="voice_minutes_few">%1$d minuty</string>
    <string name="voice_minutes_many">%1$d minut</string>
    <string name="voice_minutes_other">%1$d minut</string>

and same for _voice_hours_*_  and _voice_seconds_*_

but I am not sure when _few/many/other_ is used.  :/
Had sb enlighted me I would prepare PR

**Technical information**
 - Device: Samsung S5
 - OS: Android 6.0.1
 - Version 3.14.3  / 3823 (from f-droid)

dennisguse, 2021-02-09:
Zero/one/few/many/other is selected by `getQuantityString()`.
I have no clue about Polish, but if the (y) doesn't make any sense as a plural for minutes then kick it out :)

Relevant code is in `AnnoucementUtils.getAnnoucement()`.
Can you compile and test yourself?

matkor, 2021-02-09:
> doesn't make any sense as a plural for minutes then kick it out :)
That would be quick fix, but not final solution.

> Can you compile and test yourself?
Not yet, but I would try, if provided  any docs/tutorial for Linux.

dennisguse, 2021-02-09:
1. Get AndroidStudio: https://developer.android.com/studio
2. Then clone the repo (using AndroidStudio): https://github.com/OpenTracksApp/OpenTracks
3. Activate developer mode on your phone: https://developer.android.com/studio/debug/dev-options

Then you should be able to compile the application and start it on your phone.

matkor, 2021-02-15:
Thanks, I managed to setup, build and run modified OpenTracks app.
I am going to test it now.

Where can I find definite guide which of
`quantity=["zero" | "one" | "two" | "few" | "many" | "other"]`
is used in Polish?

To my little study, only `one/few/many` should be enough to cover all Polish grammar but I see translations for all six variants in pl/strings.xml ?

dennisguse, 2021-02-15:
Good question and I don't have a real answer.
Basically, only the ones that are actually used by a language are needed (and that depends on the actual language; my guess: zero, one, other).
Some more details can be found here: https://phrase.com/blog/posts/fundamental-android-app-localization-guide/

dennisguse, 2021-03-17:
@matkor Are you still working on this?

matkor, 2021-03-18:
> @matkor Are you still working on this?

I made changes, but had problems with testing ( I had app speaking same every second). I have seen many changes (and releases) in repo, so I will try this weekend again.

closed 2021-04-02 by dennisguse